SpinalHDL/SpinalHDL #987: bug: CI failed with "Unable to locate command >iverilog<"

Readon, 2022-12-06:
The details is log [here](https://github.com/Readon/SpinalHDL/actions/runs/3627510211/jobs/6118569392).

kleinai, 2022-12-07:
This is an odd one. Looks like your CI built(?) iverilog? The mainline CI doesn't appear to have to do that.

numero-744, 2022-12-11:
I think the source of the error is:

```
2022-12-06T09:16:18.9454195Z parse.cc:299:10: fatal error: parse.hh: No such file or directory
2022-12-06T09:16:18.9454804Z   299 | #include "parse.hh"
2022-12-06T09:16:18.9455502Z       |          ^~~~~~~~~~
2022-12-06T09:16:18.9456369Z compilation terminated.
2022-12-06T09:16:18.9478366Z make[1]: *** [Makefile:108: parse.o] Error 1
```

This might be interesting: https://github.com/steveicarus/iverilog/issues/247#issuecomment-579806944

@kleinai The CI builds iverilog here: https://github.com/SpinalHDL/SpinalHDL/blob/dev/tools.sh#L29 sourced from https://github.com/SpinalHDL/SpinalHDL/blob/dev/.github/actions/action.yml#L49

Readon, 2022-12-11:
> I think the source of the error is:
> 
> ```
> 2022-12-06T09:16:18.9454195Z parse.cc:299:10: fatal error: parse.hh: No such file or directory
> 2022-12-06T09:16:18.9454804Z   299 | #include "parse.hh"
> 2022-12-06T09:16:18.9455502Z       |          ^~~~~~~~~~
> 2022-12-06T09:16:18.9456369Z compilation terminated.
> 2022-12-06T09:16:18.9478366Z make[1]: *** [Makefile:108: parse.o] Error 1
> ```
> 
> This might be interesting: [steveicarus/iverilog#247 (comment)](https://github.com/steveicarus/iverilog/issues/247#issuecomment-579806944)
> 
> @kleinai The CI builds iverilog here: https://github.com/SpinalHDL/SpinalHDL/blob/dev/tools.sh#L29 sourced from https://github.com/SpinalHDL/SpinalHDL/blob/dev/.github/actions/action.yml#L49

Does it mean that the version of iverilog should be updated?

numero-744, 2022-12-11:
The in the iverilog issue is mentioned a fix before the release of https://github.com/steveicarus/iverilog/releases/tag/v10_3 that we seem to use so I don't think version is the cause, but that doesn't mean updating won't work!

Also, even without this error, maybe keeping our tools up-to-date is a good thing anyway?

What I don't understand is that it did not break anything on other repositories, maybe it could be a GitHub caching related?

Readon, 2022-12-12:
The errors cause by "parse.hh" is fixed by [iverilog\#349](https://github.com/steveicarus/iverilog/pull/349), which has been merged to 11_0 .

Readon, 2022-12-12:
Finally, I get those compilations passed by specifying the ubuntu-lastest to ubuntu-20.04 and update iverilog to 11_0 version.
However, I don't think it is solved, because one of the tests failed at first. I rerun it after the other get pass, it passed.

Readon, 2022-12-18:
I think this can be closed, and [this branch](https://github.com/Readon/SpinalHDL/tree/ubuntu20) can be a reference to someone met the same cases.

Readon, 2023-01-13:
I think the existing cache files work for CI. If it is not, clean the caches and rebuild it again can works, at least, it is not required each time.

numero-744, 2023-01-13:
If you clean it won't rebuild it. This solution is not clean on the long term. I'll PR something which should better handle caches, I hope this evening.

Readon, 2023-01-13:
> If you clean it won't rebuild it so this solution is not clean on the long term. I'll PR something which should better handle caches, I hope this evening.
> 
> (Currently is works only because it is in old caches which are actually outdated, so a bug makes things seem to work)

Yes, I got the reason why this is reopened. The temporarily solution is #1009

PR #1000 (2023-01-02): ci: fix missing iverilog error
Closes #987
PR #1017 (2023-01-16): ci: cache management + code organization

Closes #987

# Context, Motivation & Description

This PR addresses cache management issues we are facing.

It seems that IVerilog's build environment has issues with recent versions of Ubuntu (or its packages). Maintaining consistencies between different pieces of software and dependencies is the job of distribution maintainers so it is really simpler to just `apt install iverilog`. (Last night I have spent hours trying to make ubuntu-latest build iverilog 10.3 and 11.0, didn't find a way. Maybe it was just too late :sweat_smile: if you really really  [... cut]
closed 2023-01-16 by numero-744