SpinalHDL/SpinalHDL #999: A problem with switching SV syntax to SpinalHDL syntax

Shiyunfeng96, 2022-12-15:
Hello, I am currently developing a project based on SpinalHDL. I have a problem with syntax.

the SV:
`   wire   [7:0]  string_mode; 
if ((match_data_vld == 1'b1) && (d_set[m][string_mode[5:0]] == 1'b1){
    xxxxxxx
 }`

the SpinalHDL:
val string_mode = UInt(8 bits)
`when((match_data_vld === 1) && (d_set(m)(string_mode) == 1)
`

This seems to generate some errors
the **string_mode** cannot be applied to (Int)！

do i miss someting？

Readon, 2022-12-16:
> the SpinalHDL: val string_mode = UInt(8 bits) `when((match_data_vld === 1) && (d_set(m)(string_mode) == 1) `
                                                                                                                                                                            ^^^ Can you try to append one more "=" to here?

numero-744, 2023-01-08:
Closed as there was no answer from issue author in the last weeks. Feel free to reopen the issue if you still need help.

closed 2023-01-08 by numero-744