SpinalHDL/SpinalHDL #1003: Unable to use mixed width ram

aureoleday, 2022-12-20:
case class MemTestCfg( 
                  waddrWidth : Int = 3,
                  raddrWidth : Int = 4,
                  wdataWidth : Int = 32,
                  rdataWidth : Int = 16,
           ) {
}

case class MemTest(generics: MemTestCfg) extends Component{
  val io = new Bundle{
    val waddr = in UInt(generics.waddrWidth bits)
    val raddr = in UInt(generics.raddrWidth bits)
    val wdata = in UInt(generics.wdataWidth bits)
    val rdata = out UInt(generics.rdataWidth bits)
    val ren = in Bool()
    val wen = in Bool() 
  }

  val memCtrl = new Area{
    val mem = Mem(UInt(generics.wdataWidth bits),1<<generics.waddrWidth)
    mem.addAttribute(name="ram_style",value="block")
    mem.writeMixedWidth(io.waddr,io.wdata,io.wen)
    //io.rdata := mem.readSync(io.raddr,io.ren)
    mem.readSyncMixedWidth(io.raddr,io.rdata,io.ren)
  }
}

object MemTestCfg{
  val gen = MemTestCfg(
                  waddrWidth = 4,
                  wdataWidth = 32,
                  raddrWidth = 5,
                  rdataWidth = 16 
            )
}

// Hardware definition
object MemTestVerilog extends App {
  Config.spinal.generateVerilog(MemTest(MemTestCfg.gen))
}

the above code is my test code, that will always generate the error information below:
[error]  Verilog backend can't emit toplevel/memCtrl_mem : Mem[16*32 bits] because of its mixed width ports

how can i resolve this error?

Dolu1990, 2022-12-20:
Hi,

Mixedwidth ram are kind of a pain to infer well using regular verilog in a portable way. So instead, SpinalHDL rely on blackboxing them : 

```scala
object MemTestVerilog extends App {
  config.spinal.addStandardMemBlackboxing(blackboxRequestedAndUninferable).generateVerilog(MemTest(MemTestCfg.gen))
}
```

=> 

```verilog
  Ram_1w_1rs #(
    .wordCount(16),
    .wordWidth(32),
    .clockCrossing(1'b0),
    .technology("auto"),
    .readUnderWrite("dontCare"),
    .wrAddressWidth(4),
    .wrDataWidth(32),
    .wrMaskWidth(1),
    .wrMaskEnable(1'b0),
    .rdAddressWidth(5),
    .rdDataWidth(16)
  ) memCtrl_mem (
    .wr_clk  (clk                      ), //i
    .wr_en   (memCtrl_mem_wr_en        ), //i
    .wr_mask (1'b1                     ), //i
    .wr_addr (io_waddr[3:0]            ), //i
    .wr_data (memCtrl_mem_wr_data[31:0]), //i
    .rd_clk  (clk                      ), //i
    .rd_en   (memCtrl_mem_rd_en        ), //i
    .rd_addr (io_raddr[4:0]            ), //i
    .rd_data (memCtrl_mem_rd_data[15:0])  //o
  );
```

numero-744, 2023-01-08:
Closed as done as there was a suggestion >1 week ago without feedback from issue author. Feel free to reopen the issue if the suggestion does not match your needs.

closed 2023-01-08 by numero-744