SpinalHDL/SpinalHDL #1004: Docker-based CI GitHub workflow

numero-744, 2022-12-29:
Publishing a docker image of Spinal & tools might be interesting to make it easier for projects using SpinalHDL to build their CI workflows. I also think it would bring more flexibility, ease caching and ease local debugging.

The suggestion is to:

- Build the Docker image with tools and cache it in GitHub Workflow to keep it across runs when chosen tool versions are not updated.
- Build the Docker image `FROM` this one + `publish-local`'ing SpinalHDL and cache it to share it between jobs of the same run, bringing more flexibility without wasting processing time.
- Use the latter Docker image to run all tests with as much parallelism as possible (there is currently a little bit of parallelism but it re-builds things for each job)
- Release the latter Docker image on Spinal version releases so that it can be used by other projects' CIs.

I'll work on fixing CI issues we are facing with iverilog so I can create a Docker-based workflow if you agree with this solution. I'll begin with fixing the GitHub workflow caching of the RTD docker image.

@allexoll

wifasoi, 2022-12-29:
We were burned by the docker hub debucle.. This is why we started away from making and publishing env container. But now that GitHub does have an image hub integrated I'm not opposed to turning back.

numero-744, 2022-12-29:
@wifasoi Can you point me to documentation, please?

Btw for now I'm using https://github.com/docker/build-push-action but things after the first `ADD` (`requirements.txt` in RTD) in the `Dockerfile` are not cached in the workflow. I plan to export the image and cache it "by hand", but I think that for publishing it won't be an issue to just use `build-push-action`.

Readon, 2023-01-04:
I think it's a good idea to have a docker image that can be used in CI.
On other hand, it's really easier for new starter who can skip to set up the environment manually.
Evenmore, if it is downloadable, an offline usage can also help.

numero-744, 2023-01-04:
Ah maybe we can release a docker image in GitHub releases ^^' easier to deal with than Docker hub I guess :hand_over_mouth:

Dolu1990, 2023-01-05:
Issue is the hosting of the docker image.
Don't wan't to use a today free solution, for it to become non free the day the investors ponzy skim is done and they finaly need to make money instead of buying growth.
Already happend with travis / docker hub. It is realy wasting our time.

numero-744, 2023-01-05:
I think it can just be not released at all: since there is a Docker file, there is an easy way to get the image ready. Example: online computer clones SpinalHDL, builds the docker image and exports it to a file. This file is put into the offline computer, which can then use SpinalHDL offline. No need to install things except Docker because all the magic happens inside the image.

Also, about the last point of my suggestion, "Release the latter Docker image on Spinal version releases so that it can be used by other projects' CIs.", I think it is actually not interesting because the latter image has the context of SpinalHDL repository so it is only suitable for testing SpinalHDL itself. Another image could be made which would pre-fetche scala libraries like [SpinalNomad](https://github.com/SpinalHDL/SpinalNomad/blob/main/docker/Dockerfile) does.

Note: I have built the first image, with only tools. Not tested yet, I might build one to prepare for tests this evening, it will test the first one as it is built on it.

numero-744, 2023-01-07:
```
Step 9/11 : RUN sbt sbtVersion
 ---> Running in 2104cc4c964f
[info] [launcher] getting org.scala-sbt sbt 1.6.0  (this may take some time)...
[info] [launcher] getting Scala 2.12.15 (for sbt)...
The command '/bin/sh -c sbt sbtVersion' returned a non-zero code: 2
```

Sbt you are not helping me!

I'm getting this error in the folder where I have put SpinalHDL, but I'm not getting it in an empty folder :thinking: 

I think I'm gonna try to fix caches without Docker first.