SpinalHDL/SpinalHDL #1005: `ClockDomain.forkStimulus` on comb component yields useless error message

numero-744, 2022-12-30:
Many (all?) new spinal users are facing this error message, which is unclear. It would be nice to catch it early and print a nice error instead.

Suggestion: check that `bt` is not `null` before using it, and if it is `null` then fail with useful error message, for instance `trying to apply clock stimulus on a component without a clock`

@64

saahm, 2023-01-02:
Yea i ran into this myself aswell before asking around to use something like sleep() instead of waiting for clock edges (which makes sense for combinatorial circuits)

Maybe its more important to show/tell users how to simulate combinatorial components. I.e. since the whole clock/reset/clockdomain is hidden on purpose from the user, touching any clockdomain related things in combinatorial circuits wont make any sense anyway. 
The forkStimulus method obvious creates a clock process that runs in parallel, which will have no connection to the DUT in any sense and therefore shouldnt exist. 
For simulation wouldnt it be helpful to figure out if the user wants to progress time by sleep() (for comb) or in some other way (dependent on clockedges)?

numero-744, 2023-01-02:
True, useful explanation about this are needed in the simulation part of https://github.com/SpinalHDL/SpinalDoc-RTD/issues/151

On the other hand I think it might still be interesting to display a useful error message (at least for users who just run to coding without reading the docs ^^')

>  For simulation wouldnt it be helpful to figure out if the user wants to progress time by sleep() (for comb) or in some other way (dependent on clockedges)?

You mean just silencing the error if there is no clock pin to apply the stimulus on, and still having a fake stimulus to use `waitSampling`?

saahm, 2023-01-02:
Well maybe better: Figure out if the module is combinatorial and tell the user too? Trying to correct things a user tries to do is probably always tricky. But a good error message with maybe a hint is a good idea, i agree.

Dolu1990, 2023-01-02:
Fixed ^^
No more error should happen, as now in simulation, the toplevel clockdomain will automaticaly be pulled.

commit 096805fc (2023-01-02): Fix #1005 by automaticaly pulling the toplevel CD in a SpinalHDL phase
closed 2023-01-02 by Dolu1990