SpinalHDL/SpinalHDL #1006: The missing initialization of signals in simulation is better to be checked and warned.

Readon, 2023-01-04:
The existing simulation engine allows the presence of uninitialized signals without prompting.
The uninitialized signals will be assigned a random value determined by the seed, thus the errors will be overwritten or cause the simulation to fail.

It would be a good solution to check the uninitialized signals through the engine at the beginning of the simulation.

numero-744, 2023-01-04:
And suggest the user to use `.randomize()` to explicitly have random values ?

Dolu1990, 2023-01-04:
Uninitialized signal ? or register ?
Do you mean x-propagation ?

numero-744, 2023-01-04:
Ah I supposed it was about DUT inputs, but maybe I'm wrong

Readon, 2023-01-04:
> Ah I supposed it was about DUT inputs, but maybe I'm wrong

You're right, what I mean is the DUT's inputs.

kleinai, 2023-01-05:
I don't see this as a deficiency, more of an annoyance if you forget. If anything I'd like 9 value support at the simulation level. The red X's make it easier to see errors :P

Readon, 2023-01-06:
> I don't see this as a deficiency, more of an annoyance if you forget. If anything I'd like 9 value support at the simulation level. The red X's make it easier to see errors :P

Sometimes, the work flow is "Design->Debug with simulation->...... long time later -> Changes -> Check if it is pass". 
During this process, the wave file is not expected to be checked each time. 
However, the simulation fails not caused by the design changes but an uninitialized signal.

Dolu1990, 2023-01-16:
Could add a hash map to check that all inputs were assigned at least once, else generate a warning, but to me, it seems very specific, and shouldn't be turned on by default.

Readon, 2023-01-17:
> Could add a hash map to check that all inputs were assigned at least once, else generate a warning, but to me, it seems very specific, and shouldn't be turned on by default.

I have met this kind of problems several times, and I could not know which signals have been assigned during the simulation. Especially when xxxDriver, xxxAgent are used. I have to check the signals each time when I use it.

There are too much way to assign values to signals, so things would get complicated.

PR #1555 (2024-10-08): Watch unassigned inputs

Closes #1006 

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [x] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?
closed 2024-10-08 by Readon