SpinalHDL/SpinalHDL #1011: Remove inverter from self.ready output in s2mPipe

pavel-demin, 2023-01-10:
I would like to propose to make the code of the `s2mPipe` method in [Stream.scala](https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/Stream.scala) a little bit cleaner by replacing the `rValid` register with the `rReady` register. This way the `self.ready` output will be directly connected to the `rReady` register without an intermediate inverter and the code of this method will look slightly more similar to the code of the `m2sPipe` method.

Here is the code with the proposed changes:
```
  def s2mPipe(flush : Bool = null): Stream[T] = new Composite(this) {
    val s2mPipe = Stream(payloadType)

    val rReady = RegNextWhen(s2mPipe.ready, s2mPipe.valid) init(True)
    val rData = RegNextWhen(self.payload, self.ready)

    self.ready := rReady

    s2mPipe.valid := self.valid || !rReady
    s2mPipe.payload := Mux(rReady, self.payload, rData)

    if(flush != null) rReady.setWhen(flush)
  }.s2mPipe
```

What do you think?

likewise, 2023-01-16:
To me the existing code looks much clearer: The ready becomes valid when rData is free, thus when it does not contain valid data already. This is exactly what the existing code says.

What I think is needed are good code comments.

pavel-demin, 2023-01-18:
Thank you for your comment.

What is confusing to me about the current `s2mPipe` code is that normally I would expect a register for an output signal to have the same name or a similar name to the output signal it is connected to.

For example, in the `m2sPipe` method, the valid output register is called rValid, the payload output register is called rData.

I think it would be more consistent if the ready output register was called rReady.

I also think that in some cases it is important that an output is connected directly to a register. I think if there is an inverter between a register and an output it could be confusing for synthesis tools and not all tools would be able to optimize this inverter out.

likewise, 2023-01-22:
I think the current code is perfect and in good relationship to common design methods for a skid buffer.

See https://zipcpu.com/blog/2019/05/22/skidbuffer.html

```
always @(*)
		o_ready = !r_valid;
```
and see (t)his remark: "I built and implemented many [skid buffer](https://zipcpu.com/blog/2017/08/14/strategies-for-pipelining.html)s before realizing this. Even once I first saw this equivalency, it still took some time (and [formal](https://zipcpu.com/blog/2017/10/19/formal-intro.html) proofs) to believe it. That said, it nicely simplifies any implementation."

So I counter your proposal. I would prefer the code to remain as-is.

likewise, 2023-01-22:
The "equivalency" Dan Gisselquist mentions is that the s2mPipe() is ready to accept new data when the internal data register does not contain valid data. This condition is equivalent and thus
```
o_ready = !r_valid;
```
or
```
    self.ready := !rValid
```
contains this equivalence perfectly.

The rValid can be seen as the internal state, and ready is the output. I also disagree that the output signal name should match an internal register state name.

pavel-demin, 2023-01-22:
Looks like I was not clear enough in my previous comments. I am sorry. Let me try again.

The main problem for me is that in the current code there is an inverter between the `rValid` register and the `self.ready` output. My main suggestion is in the title: "Remove inverter from self.ready output in s2mPipe".

I think that it is good practice to have a registered output if possible. An inverter between a register and an output does not seem like the best possible implementation.

I have read several articles from the ZipCPU blog and found that a better and cleaner implementation of skid buffers is hidden in an earlier article at [this link](https://zipcpu.com/blog/2019/01/12/demoaxilite.html). The skid buffers in this article have registered outputs without inverters.

likewise, 2023-01-22:
Ah, I see, my fault at not recognizing your improvement suggestion. So basically you would like (I assume all) the outputs to be registered, so that the combinational signal delay is hidden by pure registered-only-outputs.

Then we have two more outputs to care about:
```
    s2mPipe.valid := self.valid || rValid
    s2mPipe.payload := Mux(rValid, rData, self.payload)
```
And so we should look if we could improve this design.
I think these are equal or worse than the inverter, assuming LUT3-6 and MUX building blocks.

pavel-demin, 2023-01-22:
It is OK for `s2mPipe.valid` and `s2mPipe.payload` to be not registered in the `s2mPipe` method. The `valid` and `payload` outputs are registered in the `m2sPipe` method. If registers on both `m` and `s` sides are required, then `s2mPipe` and `m2sPipe` can be chained together to create such a configuration as it is already done in the `<-/<` and `>/->` methods.

likewise, 2023-01-22:
Agreed on `s2mPipe` and `m2sPipe`. I'll continue to ask dumb questions:

I understand all your code changes easily except this change, I would expect `setWhen` would change into `clearWhen` and vice versa.
```
    // 
    val rValid = RegInit(False) setWhen(self.valid) clearWhen(s2mPipe.ready)
```
```
    // if internal data register is full, become ready after it is read
    val rReady = RegNextWhen(s2mPipe.ready, s2mPipe.valid) init(True)
```

pavel-demin, 2023-01-22:
I agree that this is the least obvious of the changes. It is not a direct transformation. It is a simplification.

The only case where `rReady` needs to be updated is when `s2mPipe.valid` is high. If  `s2mPipe.valid` is low, then we do not care about `s2mPipe.ready`.

It is similar to the logic around `rValid` in the `m2sPipe` method but with inverting and swapping of the inputs:

`rValid` in `m2sPipe`:
```
val rValid = RegNextWhen(self.valid, self.ready) init(False)
```
`rReady` in `s2mPipe`:
```
val rReady = RegNextWhen(s2mPipe.ready, s2mPipe.valid) init(True)
```
This logic passes all the functional and formal tests I have thrown at it. Including formal tests provided by ZipCPU for the AXI4-Lite bus.

likewise, 2023-01-29:
Although I was pessimistic at first, this indeed might be an improvement. It would be nice to have formal checks included.

Do we know if the synthesis results in a registered output (instead of the inverter sitting behind a register)?

pavel-demin, 2023-01-29:
> Do we know if the synthesis results in a registered output (instead of the inverter sitting behind a register)?

The following screenshots show what Xilinx Vivado generates for the current version of the `s2mPipe` method with unregistered `self.ready` output and for the new version with registered `self.ready` output listed in my first comment.

For the current version of the `s2mPipe` method, Xilinx Vivado places a single input LUT between the `rValid` register and the `self.ready` output.

For the new version of the `s2mPipe` method, Xilinx Vivado generates a registered output.

- current version with unregistered `self.ready` output
[image]

- new version with registered `self.ready` output
[image]

Dolu1990, 2023-06-12:
I think this may have been in since : 
https://github.com/SpinalHDL/SpinalHDL/commit/180eabc827e49ba58d55115b8c21cddc0b08716f

pavel-demin, 2023-06-12:
Great! Thank you!

closed 2023-06-12 by pavel-demin