SpinalHDL/SpinalHDL #1012: Dependency Review and Upgrade

kleinai, 2023-01-10:
I'd like to request, and help preform, a review of the dependencies and upgrade them as necessary. I'd like to try to use Spinal at my office and they would prefer more up-to-date libraries than what are being used now.

kleinai, 2023-01-10:
This is what we're currently using unless I missed anything.

Org | Package | Current Version | Date |   | Latest Version | Date
-- | -- | -- | -- | -- | -- | --
com.github.oshi | oshi-core | 5.2 | 26-Jun-20 |   | 6.4.0 | 3-Dec-22
com.github.scopt | scopt | 3.7.1 | 13-Jun-19 |   | 4.1.0 | 2-Jul-22
com.lihaoyi | sourcecode | 0.2.7 | 13-May-21 |   | 0.3.0 | 8-Jul-22
commons-io | commons-io | 2.4 | 12-Jun-12 |   | 2.11.0 | 13-Jul-21
net.java.dev.jna | jna | 5.5.0 | 30-Oct-19 |   | 5.12.1 | 29-Jun-22
net.java.dev.jna | jna-platform | 5.5.0 | 30-Oct-19 |   | 5.12.1 | 29-Jun-22
net.openhft | affinity | 3.21ea1.1 | 26-May-21 |   | 3.23.2 | 29-Sep-22
org.scala-lang.modules | scala-xml | 1.2.0 | 7-Jun-19 |   | 2.1.0 | 4-Apr-22
org.scalatest | scalatest | 3.2.5 | 19-Feb-21 |   | 3.2.15 | 6-Jan-23
org.slf4j | slf4j-api | 1.7.25 | 16-Mar-17 |   | 2.0.6 | 13-Dec-22
org.slf4j | slf4j-simple | 1.7.25 | 16-Mar-17 |   | 2.0.6 | 13-Dec-22

Possible upgrade versions listed. Of course some of these are probably API incompatible will have to be individually checked.

Readon, 2023-01-11:
I have experienced terrible struggles with library upgrades.
Sometimes "latest" means changes, which need to pay a lot of attention to fix it right.

numero-744, 2023-01-11:
Maybe a good start can be to fix all deprecation warnings that are displayed (sometimes only counted as they are really a lot) while compiling SpinalHDL?

kleinai, 2023-01-12:
Almost all of those warnings are within our code and not against the dependencies. That would be a good topic for a different clean up issue.

PR #1016 (2023-01-13): build: upgrade dependency versions

Closes #1012

Flat upgrading all the dependencies seems to have worked well with only one incompatibility.

# Context, Motivation & Description

Upgrade dependencies to keep current.

# Impact on code generation

None expected.
closed 2023-01-13 by numero-744