SpinalHDL/SpinalHDL #1014: add loadHex function for SparseMemory

hanm2019, 2023-01-12:
how about add a `loadHex` function for SparseMemory in `lib/src/main/scala/spinal/lib/bus/amba4/axi/sim/AxiMemorySim.scala`, so that the memory can be inited both in bin and hex file.

Dolu1990, 2023-01-26:
Sure why not ^^

Just one thing which is weird, is that there is currently two implementation of SparseMemory XD

hanm2019, 2023-01-26:
Yes, I will try to merge them.

hanm2019, 2023-01-28:
@Dolu1990 When I merge these two SparseMemory, I found that the [SpinalSimBmbLengthSpliterTester](https://github.com/SpinalHDL/SpinalHDL/pull/1015/commits/ad52b699452ad294630cd24e6a890c5a64d5cedf#diff-96ffba064b7bdc1112fd13c0756a0cef50d7b8d58f6125866a929ea28252b5b5L8) met some problem. I have pick them out. 

https://github.com/SpinalHDL/SpinalHDL/blob/ad52b699452ad294630cd24e6a890c5a64d5cedf/tester/src/test/scala/spinal/tester/scalatest/SpinalSimBmbLengthSpliterTester.scala#L10-L64

1. In the case named `failed to stop case: only read `, the SparseMemory raises several warning with `Page fault while reading page 0 (0x0)` which is because the the read address has never write data. and this case can not stop.
2. In the case named `bypass_wr`, the testbench also write after read to the memory, and raises an Error on the `onRspRead` funtion.