SpinalHDL/SpinalHDL #1020: submodule name error when multi inst

aiit-linry, 2023-01-17:
[image]
as shown in the figure, I inst 4 MyTopLevel in top module, the name of submodule in generated code should be MyTopLevel instead of MyTopLevel_3. I think its a bug.

Readon, 2023-01-17:
emm, there are cases that user initiate components in for loop with different arguments. In that case, an individual component for each is needed.

aiit-linry, 2023-01-17:
> emm, there are cases that user initiate components in for loop with different arguments. In that case, an individual component for each is needed.

this case in 1.7.3a is ok, generate 4 same "MyTopLevel". but in 1.8.0, it generate 4 same "MyTopLevel_3"

Readon, 2023-01-17:
> > emm, there are cases that user initiate components in for loop with different arguments. In that case, an individual component for each is needed.
> 
> this case in 1.7.3a is ok, generate 4 same "MyTopLevel". but in 1.8.0, it generate 4 same "MyTopLevel_3"

Then that would caused by some generating logic. @Dolu1990

numero-744, 2023-01-17:
Recent behavior change in merging in generation, maybe related to https://github.com/SpinalHDL/SpinalHDL/pull/984

Dolu1990, 2023-01-17:
Fixed ^^

aiit-linry, 2023-01-29:
> Fixed ^^

[image]
@Dolu1990 not fixed yet, name of MyTopLevel_3 still exists, but comment has fixed

aiit-linry, 2023-01-29:
> > Fixed ^^
> 
> [image] @Dolu1990 not fixed yet, name of MyTopLevel_3 still exists, but comment has fixed

I think it should be "MyTopLevel_1/2/3 replaced by MyTopLevel", and name of module is "MyTopLevel"

aiit-linry, 2023-01-29:
reopen this issue please

Dolu1990, 2023-01-30:
Should be all good now, got : 

```
//StreamFifo_2 replaced by StreamFifo

//StreamFifo_1 replaced by StreamFifo

module StreamFifo (
```

pthuang, 2024-08-22:
@Dolu1990 not fixed yet, name of MyTopLevel_3 still exists in Spinal1.10.2
The module combine_2ch_4 should actually be named combine_2ch
[image]

Dolu1990, 2024-09-03:
Hi @pthuang,

Can you provide a self contained example ?

You can use \`\`\`scala code code code \`\`\` to preserve the formating.

Thanks ^^

pthuang, 2024-09-04:
@Dolu1990 Sure, here's an example of my design:
First of all, I designed a two-channel combiner module which is very common in the communication system. On this basis, I designed a 4-channel combiner module. In the 4-channel combiner, I instantiated three two-channel combiner modules.
However, when I designed an 8-channel combiner, in this 8-channel combiner, I instantiated two 4-channel combiner and one 2-channel combiner. In the verilog file generated after compilation, the **_4** was added to the file name of the most basic 2-channel combiner module, which was not what I expected.
Here is some of the code：
```scala
case class combine_2ch() extends Component{
    val io = new Bundle{
        val combine_mode = in Bits(2 bits)
        val din = Vec(slave Flow(Bits(32 bits)), 2)
        val dout = master Flow(Bits(32 bits))
    }
    noIoPrefix()
    ......
}

case class combine_4ch() extends Component {
    val io = new Bundle {
        val combine_mode = in Bits (4 bits)
        val din = Vec(slave Flow (Bits(32 bits)), 4)
        val dout = master Flow (Bits(32 bits))
    }
    noIoPrefix()
    val comb_cas0 = for (idx <- 0 until(2)) yield {
        val combine_2ch = new combine_2ch().setName(s"comb_cas0_$idx")
        combine_2ch
    }
    val comb_cas1 = new combine_2ch()
    for (idx <- 0 until(2)) {
        comb_cas0(idx).io.combine_mode <> io.combine_mode(2*idx+1 downto(2*idx))
        for (j <- 0 until(2)) {
            comb_cas0(idx).io.din(j) <> io.din(2*idx+j)
        }
    }
    
    comb_cas1.io.combine_mode <> (io.combine_mode(3 downto(2)).orR ## io.combine_mode(1 downto(0)).orR)
    for (idx <- 0 until(2)) {
        comb_cas1.io.din(idx) <> comb_cas0(idx).io.dout
    }
    io.dout <> comb_cas1.io.dout
}

case class combine_8ch() extends Component {
    val io = new Bundle {
        val combine_mode = in Bits (8 bits)
        val din = Vec(slave Flow (Bits(32 bits)), 8)
        val dout = master Flow (Bits(32 bits))
    }
    noIoPrefix()

    val comb_cas0 = for (idx <- 0 until(2)) yield {
        val combine_4ch = new combine_4ch().setName(s"comb_cas0_$idx")
        combine_4ch
    }
    val comb_cas1 = new combine_2ch()

    for (idx <- 0 until(2)) {
        comb_cas0(idx).io.combine_mode <> io.combine_mode(4*idx+3 downto(4*idx))
        for (j <- 0 until(4)) {
            comb_cas0(idx).io.din(j) <> io.din(2*idx+j)
        }
    }

    comb_cas1.io.combine_mode <> (io.combine_mode(7 downto(4)).orR ## io.combine_mode(3 downto(0)).orR)
    for (idx <- 0 until(2)) {
        comb_cas1.io.din(idx) <> comb_cas0(idx).io.dout
    }
    io.dout <> comb_cas1.io.dout
}

object RtlGen extends App {
    SpinalConfig.generateVerilog(new combine_8ch()).printPruned()
}
```
[image]
[image]

Dolu1990, 2024-09-11:
Should be good now ^^

pthuang, 2024-09-11:
I know, but have not tried, my development environment is win11+IDEA, will not use the specified branch/node dependencies, looking forward to the subsequent official version.

commit 7480508b (2023-01-17): Fix #1020
commit ee2c4214 (2023-01-30): fix #1020 Component.definitionName are now allocated by component child order
commit 1c25b6c4 (2024-09-09): Fix #1020 Components are now named in a more preserved order