SpinalHDL/SpinalHDL #1022: use VCS+AxiMemorySim cause vpi error

aiit-linry, 2023-01-17:
[image]

Readon, 2023-01-17:
I think it is similar with #973 where you can try "sbt clean" and rebuild again.

aiit-linry, 2023-01-17:
> I think it is similar with #973 where you can try "sbt clean" and rebuild again.

I have tried “sbt clean” but still exist

aiit-linry, 2023-01-17:
> I think it is similar with #973 where you can try "sbt clean" and rebuild again.

it appears in 1.8.0, in 1.7.3a its works ok

Readon, 2023-01-17:
Yes, the get_time_precision is merged in 1.8.0. 

merlionfire has describe what he/she did in gitter.
```
 I am supposed to clean first otherwise .o is still old and doses NOT contain the expected native functions like get_time_precision*.

Eventually, I can run single-step VCS compilation based on local published version and succeed in simulation with fsdb file dumped.
```

aiit-linry, 2023-01-17:
@numero-744 no, I have deleted all "*.o" file manually in my project and excuted "sbt clean", this question still here

Readon, 2023-01-17:
> @numero-744 no, I have deleted all "*.o" file manually in my project and excuted "sbt clean", this question still here

How about clear the simulation dir, like simWorkspace?

aiit-linry, 2023-01-17:
> > @numero-744 no, I have deleted all "*.o" file manually in my project and excuted "sbt clean", this question still here
> 
> How about clear the simulation dir, like simWorkspace?

done! Thats amazing, sorry for my stupid.

closed 2023-01-17 by aiit-linry

Readon, 2023-01-17:
> > > @numero-744 no, I have deleted all "*.o" file manually in my project and excuted "sbt clean", this question still here
> > 
> > 
> > How about clear the simulation dir, like simWorkspace?
> 
> done! Thats amazing, sorry for my stupid.

We should add a tip on FAQ to Documentation, where the first steps suggested to people is "sbt clean" and "remove simulation directory" when people met errors with spinal.sim.vpi.

I just find the solution by review the code for vpi wrappers' generation.