SpinalHDL/SpinalHDL #1024: The AxiLite4 RegIf might fail on b channel without always-true ready signal.

Readon, 2023-01-18:
I have modified the AxiLite4Driver with a randomized ready signal for b channel.
Then the test fails, as [this CI job](https://github.com/Readon/SpinalHDL/actions/runs/3941305192/jobs/6743767808) shows.

I think the reason is that there is only one pulse of valid signal for b channel, which could not entering a stall status. 
And according to the spec of Axi4Lite, b channel should act as a normal stream, supporting stall.

I did not test the r channel, it might be acting similarly too.

@jijingg

Dolu1990, 2023-02-06:
@Readon Can you provide a way to reproduce the issue ? (git branch or self contained example)

Readon, 2023-02-06:
> @Readon Can you provide a way to reproduce the issue ? (git branch or self contained example)

Yes, there is a branch on my repository called regif-test. where i am also tried to fix that.

Dolu1990, 2023-02-06:
What would be the SBT command to run the failing test ?

Readon, 2023-02-06:
> What would be the SBT command to run the failing test ?

The test would not end while run ```sbt test```.

Dolu1990, 2023-02-06:
> sbt test

Issue with it is that it involve make an ocean of other things. Could you point the specific scala test failing ?

(testOnly xxx.yyy.zzz)

Readon, 2023-02-06:
> > sbt test
> 
> Issue with it is that it involve make an ocean of other things. Could you point the specific scala test failing ?
> 
> (testOnly xxx.yyy.zzz)

``` sbt "testOnly *.RegIfAxiLite4Tester" ``` would trigger that.

Dolu1990, 2023-02-06:
Thanks :D

Dolu1990, 2023-02-09:
should be good now

commit 451f74fb (2023-02-09): Fix #1024 RegIfAxiLite4Tester and AxiLite4BusInterface backpresure
closed 2023-02-09 by Dolu1990