SpinalHDL/SpinalHDL #1027: Add opt-in feature for Spinal ".mux" to generate parallel_case

azaparov, 2023-01-22:
When mux is fully specified without masked bits it would be good to have option for the spinal to generate "parallel_case" directive in the produced HDL.
I.e 
val foo = UInt(2 bit)
val muxed = io.foo.mux(B"00" -> U(4, 4 bit), B"01" -> U(2, 4 bit), B"10" -> U(4, 4 bit),  B"11" -> U(1, 4 bit))
should produce case with "parallel_case", most likely should be opt-in option such as additional parameter to "mux" or possibly "parMux"

Dolu1990, 2023-01-26:
I think the idea is that if the synthesis see a fully covered case, he will be able by himself to generate the optimal logic, parallel_case being more for the user to specify stuff like onehot signal having no overlapping in a switch case ?

numero-744, 2023-01-31:
@azaparov Is the reply above okay for you?

azaparov, 2023-02-09:
I would agree, we need general way to specify parallel conditional block, tools are not always do what designer wants.