SpinalHDL/SpinalHDL #1029: Bug: Wrong path for .bin, .v and .vhd files during 'sbt test'.

Readon, 2023-01-31:
When running
``` bash
sbt test
```
for the library while developing.

files like RomTester3.v_toplevel_rom_symbol0.bin would be placed in project directory.
and "rm: cannot remove 'tester/src/test/python/spinal/RomTester3/RomTester3.v_toplevel_rom_symbol0.bin': No such file or directory" is reported.

There are also many .v and .vhd files in the project's root.
I guess the new CI test code does not solve the path for generated files.

@numero-744

numero-744, 2023-01-31:
I don't really understand the issue. You `cd` in the root folder of `SpinalHDL` and run `sbt test` and get an `rm` error?

I think there is no relationship with CI, but I need more information to be sure.

Readon, 2023-02-01:
> I don't really understand the issue. You `cd` in the root folder of `SpinalHDL` and run `sbt test` and get an `rm` error?
> 
> I think there is no relationship with CI, but I need more information to be sure.

emm, the best way is to run it at local.
You can notice that many generated files like .v and .vhd are in the root directory of project, where they are normally in the simWorkspace directory.

Readon, 2023-02-01:
@numero-744 sorry, I have confirm that even in v1.7.3a some generated verilog and vhd files are still placed in root directory of the SpinalHDL project.

numero-744, 2023-02-01:
Okay I better understand now. So actually I have always got that behavior on my side; by default Spinal generates files in the root folder. This is why I have added a `src/gen/` folder in the template and the appropriate configuration so that generated files go in it.

I understand that generating in a folder would be better, now I think the questions are:

- Where to put the these files?
- How to enforce that this choice is used on all tests that generate files (without simulation, in this case there is a specific folder)

Btw I had suggested that we could add a way to perform design checks without generating a file, but it didn't interest people.

andreasWallner, 2023-02-11:
What about changing the default of where files are generated, similar to how `SpinalSimConfig` uses `simWorkspace` by default?
(From what I see most of the files generated are for cocotb tests...?)

Readon, 2023-02-11:
> What about changing the default of where files are generated, similar to how `SpinalSimConfig` uses `simWorkspace` by default? (From what I see most of the files generated are for cocotb tests...?)

Is that related with the implementation of SpinalTesterCocotbBase?

andreasWallner, 2023-02-12:
> 

I'll have a look

andreasWallner, 2023-02-13:
@Readon I had a quick look and yes, those files are generated from the Cocotb test cases.
Since I couldn't get cocotb to work on my machine I opened #1036 just as a test vehicle to see if only those changes there would be sufficient to move the output.

PR #1053 (2023-03-21): Retide the generated files by tester/test cases.

Closes #1029 

# Context, Motivation & Description

1. Avoid to generate intermediate files in the root directory while running "sbt test".
2. Move temp files for cocotb into individual cocotbWorkspace directory.
3. Create a SpinalAnyFunSuite to wrap all spinal tester specific default settings.
4. Expose SpinalTesterCocotbBase and SpinalTesterGhdlBase to user who can inherit and use it if they want to use cocotb and ghdl correspondingly.

A lot of changes are based on @andreasWallner 's PR #1036. Thanks a lot.

# Impact on code generation

None [... cut]
closed 2023-03-21 by Readon