SpinalHDL/SpinalHDL #1030: Regif: WishboneBusInterface writeData is driven by bus.DAT_MISO instead of bus.DAT_MOSI

boehmerst, 2023-02-02:
I am exploring the register interface builder to generate a small register bank and was wondering why any register write operations seem not to work. From the generated Verilog code I can see that the registers are driven by the module output. Patching the generated code resolves the issue.

In the WishboneBusInterface of the register interface builder the _writeData_ is driven by _bus.DAT_MISO_. I assume it should be _bus.DAT_MOSI_.

```
bus.ACK := True
bus.DAT_MISO := readData

val selMatch = if (bus.config.useSEL) bus.SEL(selId) else True
val askWrite = (selMatch && bus.CYC && bus.STB && bus.WE).allowPruning()
val askRead = (selMatch && bus.CYC && bus.STB && !bus.WE).allowPruning()
val doWrite =
  (selMatch && bus.CYC && bus.STB && bus.ACK && bus.WE).allowPruning()
val doRead =
  (selMatch && bus.CYC && bus.STB && bus.ACK && !bus.WE).allowPruning()
val writeData = bus.DAT_MISO // <- should be bus.DAT_MOSI right?
```

andreasWallner, 2023-03-05:
@Dolu1990 With you fix we should be able to close this?

Dolu1990, 2023-03-07:
Hoo right, forgot about it, yes ^^

closed 2023-03-07 by Dolu1990