SpinalHDL/SpinalHDL #1031: ram_style default attribute is causing problems when using Yosys

wel97459, 2023-02-03:
Yosys is throwing up an error `ERROR: no valid mapping found for memory` when it sees `(* ram_style = "distributed" *)`

I think that it would be a better solution to document the configuration `device = Device.XYZ` then to have the default make Yosys throw up on newbie projects.

But even device as the config flag here is not the right naming, it should be something like `toolChain = Tool.XYZ`    

https://github.com/SpinalHDL/SpinalHDL/blob/ad62c4ea3fc0acf70885441a88fe8f2336e4cb20/core/src/main/scala/spinal/core/internals/Phase.scala#L1230

Thanks,

qbg, 2023-02-08:
I also ran into issues with `ram_style` using the Intel toolchain. It would be useful to at least have about `device` on the docs page for [RAM/ROM](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Sequential%20logic/memory.html).

Dolu1990, 2023-02-09:
Hi,

> so assuming one tool as a default maybe isnt good either 

I totaly agree.

But, here is a very tricky thing, basicaly, the idea of this wasn't to enforce (* ram_style = "distributed" *), but instead to avoid the xilinx tool using block ram which would break the described read during write behaviour.

So, basicaly, the bug of the Xilinx tool is that using block ram for an assyncronus read, the latency from write to read the same address it is increase by one cycle,  which break the wished behaviour.

In many ways, i prefer a tweeked default config which may trigger explicit errors on some tools, than silent ones on some others. As it will likely save time for the most people (utilitarian mindset), especialy as that silent error can only be seen after the tool syntesis, i will make people lose days finding it

Maybe one potential alternative to this could be to have some define / ifdef in the generated verilog which detect when we are on xilinx tools ? That would be cleaner i think.

> I also ran into issues with ram_style using the Intel toolchain. It would be useful to at least have about device on the docs page for [RAM/ROM](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Sequential%20logic/memory.html).

@qbg   Which issues ?

qbg, 2023-02-09:
It generates the warning `Warning (10335): Unrecognized synthesis attribute "ram_style"`. Trying it again, it appears to synthesize despite that though.

Readon, 2023-02-09:
> Hi,
>
>
> I totaly agree.
>
> But, here is a very tricky thing, basicaly, the idea of this wasn't to enforce (* ram_style = "distributed" *), but instead to avoid the xilinx tool using block ram which [...]

Maybe an individual phase to deal with the vendor dependent backend is a way to help.

wel97459, 2023-02-09:
Well on the flip side is to have Yosys throw a warning, and not bomb with an error.

saahm, 2023-02-10:
But isnt the addition of that attribute not a bit a user-freedom? I even within the same design I can use Mem() in various spots and sometimes want BRAM and sometimes not. And sometimes the BRAM behaves one way (e.g. the Xilinx way) and sometimes it behaves the way SpinalHDL expects.

Dolu1990, 2023-02-10:
@saahm 

> But isnt the addition of that attribute not a bit a user-freedom? I even within the same design I can use Mem() in various spots and sometimes want BRAM and sometimes not. And sometimes the BRAM behaves one way (e.g. the Xilinx way) and sometimes it behaves the way SpinalHDL expects.

That attribute is only added when the ram has asyncronous read, which is in itself the explicitly choice made by the user i would say.

If you use syncRead, it won't add the attribut.

> Maybe an individual phase to deal with the vendor dependent backend is a way to help.

Yes, that's the idea as soon as you need something realy specific.

wel97459, 2023-02-10:
In my case I'm only using the mem as a rom and I don't  have a write statement.

Dolu1990, 2023-02-13:
> In my case I'm only using the mem as a rom and I don't have a write statement.

@wel97459 Ahhh, yes right. I just pushed a fix for that case.

wel97459, 2023-02-14:
Oh awesome thanks

closed 2023-02-14 by wel97459