SpinalHDL/SpinalHDL #1040: Verilator on FreeBSD: Incorrect include path

Tectu, 2023-02-21:
When building a simulation using the Verilator backend, an include path for JNI are incorrect on FreeBSD.
Using [java/openjdk8](https://www.freshports.org/java/openjdk8/), the path `/usr/local/openjdk8/include/linux` does not exist. Instead, it is `/usr/local/openjdk8/include/freebsd`.

Using the following configuration leads to a successful build:
```scala
SimConfig.withVerilator.addSimulatorFlag("-I/usr/local/openjdk8/include/freebsd").compile({}).doSim({})
```

Big thanks to @dlmiles for helping out on gitter to "resolve" this problem.

dlmiles, 2023-02-22:
I should have a patch for this one the way within a week with other Verilator backend improvements.

Tectu, 2023-06-23:
@dlmiles any news on this? :)

Dolu1990, 2023-07-31:
@dlmiles With Tectu, we made a screen sharing party, got it fix :)

commit 67462c1b (2023-07-31): fix #1040 Merge branch 'freebsd' into dev