SpinalHDL/SpinalHDL #1042: Plic issue

chenbo-again, 2023-02-22:
hi, I wonder what does that mean.
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/misc/plic/PlicMapper.scala#L121

why behavior of plic has connection with bus transaction on all address?

chenbo-again, 2023-02-22:
and in naxriscv(it use axilite directly)
```
    bus.onReadPrimitive(AllMapping, haltSensitive = false, documentation = "")
```

this will be omit, because of do nothing to `AllMapping` https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axilite/AxiLite4SlaveFactory.scala#L63

Was it intentional?

Dolu1990, 2023-03-18:
If i remember well, that counter is there to allow some pipelining into the internal logic of the plug.
See https://github.com/SpinalHDL/SpinalHDL/commit/56b89bceb6f1e58c852bf0b9d9117b22e499f5cb#diff-4526764bfc0477013637889bac4fe8b195e4f17b226e8cdab5e985150ee556c3R36
Sorry for the delay :/

closed 2023-03-30 by chenbo-again