SpinalHDL/SpinalHDL #1049: subdivideIn function gets wrong width of the slices

01xz, 2023-03-05:
hi, i tried to split the `val i = Bits(8 bits)` with `i.subdivideIn(5 bits, false)`, 
which i was expecting to get a `Vec` like this: `Vec(Bits(5 bits), Bits(3 bits))`.

but the SpinalHDL will actually split it into `Vec(Bits(4 bits), Bits(4 bits))`.

here is the code i tested:

```scala
case class Test() extends Component {
  val io = new Bundle {
    val i = in Bits(8 bits)
    val o = out Vec(Bits(5 bits), 2)
  }
  val temp = Vec(Bits(5 bits), Bits(3 bits))
  val tmp = io.i.subdivideIn(5 bits, false)
  io.o(0) := tmp(0).resized
  io.o(1) := tmp(1).resized
}
```

and the generated `.v` file:

```verilog
// Generator : SpinalHDL v1.8.0b    git head : 761a30e521263983ddf14de3592f7a9f38bf0589
// Component : Test
// Git hash  : 98857617cfac48aac841f1f5f0fd2e4e33b6b0d3

`timescale 1ns/1ps

module Test (
  input      [7:0]    io_i,
  output     [4:0]    io_o_0,
  output     [4:0]    io_o_1
);

  wire       [4:0]    temp_0;
  wire       [2:0]    temp_1;
  wire       [3:0]    tmp_0;
  wire       [3:0]    tmp_1;

  assign tmp_0 = io_i[3 : 0]; // @ BaseType.scala l299
  assign tmp_1 = io_i[7 : 4]; // @ BaseType.scala l299
  assign io_o_0 = {1'd0, tmp_0}; // @ CodeGen.scala l31
  assign io_o_1 = {1'd0, tmp_1}; // @ CodeGen.scala l32

endmodule
```

it seems that in [https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/BitVector.scala#L322](https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/BitVector.scala#L322), 
only calculate the `sliceCount` is not enough to get the correct width of the slice, 
and will split the 8 bits to two 4 bits in my situation.

Readon, 2023-03-05:
I think the default behaviour of subdivideIn is to divide signal into equal parts. 
so, if you want a function for specific usage, it should be implemented by your own.
it would not that hard to write a loop that extract sub signals in specified width array.

andreasWallner, 2023-03-05:
@Readon I think this is a Bug, looking at the `subdivideIn` docs/signature (https://github.com/SpinalHDL/SpinalHDL/blob/2867fa8fdf893f589f6303dbf556146afb10c864/core/src/main/scala/spinal/core/BitVector.scala#L307 and the following function) there are both versions, one that splits into pieces of n bits, the other that splits into n slices.

Testing this a bit (and looking at the code) it works except in cases where the number of slices that is calculated in `subdivideIn(BitCount)` could cleanly divide the `BitVector`.

```scala
case class TestX() extends Component {
  val io = new Bundle {
    val i = in Bits(9 bits)
  }
  val temp = Vec(Bits(5 bits), Bits(3 bits))
  for (i <- 1 to 8) {
    println(i)
    val tmp = io.i.subdivideIn(i bits, false)
    println(tmp.map(t => t.getWidth).mkString(" "))
  }
}
object TestXX extends App {
  SpinalVerilog(TestX())
}
```
Will print
```
1 1 1 1 1 1 1 1 1
2
2 2 2 2 1
3
3 3 3
4
3 3 3
5
5 4
6
5 4
7
5 4
8
5 4
```
Also looking at the usages in the SpinalHDL lib, I'd not expect this behavior.

PR #1051 (2023-03-18): Fix subdivideIn(..., n bit) for corner cases
When splitting 8 into 5 the old algorithm would generate an inconsistent result with other cases as it would generate two slices of 4 bit. In most other cases subdivideIn would generate the expected width as long as possible, and have a smaller last element.

If this is actually expected behavior I'd update the documentation to mention it.

Closes #1049 

# Impact on code generation

None

# Checklist

- [x] Unit tests were added
closed 2023-03-18 by Dolu1990