SpinalHDL/SpinalHDL #1050: Unexpected outcome with `subdivideIn(UInt(9 bit), 4 slices)`

andreasWallner, 2023-03-05:
When looking in detail at #1049 I found a case with kind of unexpected behavior:
```scala
val i = Bits(9 bit)
val split = i.subdivideIn(4 slices, strict = false)
```
Does generate 4 slices, but the last slice is 0-sized. In this case we could argue that it did not really split the vector in 4 pieces?
Do we expect this or is this a bug?
To really split into 4 slices `subdivideIn` would have to produce slices with size of 2, 2, 2, 3 bit or 3, 2, 2, 2 bit ...
Not sure though which is the version I'd most expect (3,2,2,2 most likely?) - in any case we should either fix this or I'd add a note to the documentation hinting that this is to be expected...

saahm, 2023-03-08:
I only have some questions for further discussion, am not really fixed to one or the other way of handling this.

Would it make sense to divide a 9 bit value into an even number of slices?
I'd expect some mathematical property to cover most cases, something the slice number is a integer divisor of the bitsize, dividing wouldnt make much sense otherwise no?
I cannot think of a use case where uneven slicing would make that much sense (maybe this is a case for having the property as an assert unless the user really wants it (and knows what he is doing)? If there is a usecase where even the (2,2,2,3) or other slicings are necessary. Maybe something DSP related (as i remember there were already things like 24 and 48 bit things going on)?

andreasWallner, 2023-03-29:
After thinking about it some more I'm not convinced that there really is a use-case for subdivideIn - closing this for now, but will add a note to the documentation.

closed 2023-03-29 by andreasWallner