SpinalHDL/SpinalHDL #1054: AxiLite4Driver simluation error in v.18.0 or later

aureoleday, 2023-03-07:
the AxiLite4Driver works fine in 1.7.3 but not in 1.8.0

my code is:
`info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[info] For instance :
[info]   val result = Bool()
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[error] Exception in thread "main" java.lang.NullPointerException
[info]   val a,b = Bool()
[info]           
[error]         at spinal.core.sim.package$.setLong(package.scala:143)
[error]         at spinal.core.sim.package$SimBoolPimper.$hash$eq(package.scala:329)
[error]         at spinal.lib.sim.StreamDriver.<init>(Stream.scala:94)
[error]         at spinal.lib.sim.StreamDriver$.apply(Stream.scala:57)
[error]         at spinal.lib.bus.amba4.axilite.sim.AxiLite4Driver.<init>(AxiLite4Driver.scala:41)
[error]         at datapath.DataPathTb.<init>(DataPathSim.scala:9)
[error]         at datapath.DataPathSim$.$anonfun$main$1(DataPathSim.scala:67)
[error]         at spinal.core.internals.PhaseCreateComponent.$anonfun$impl$262(Phase.scala:2435)
[error]         at spinal.core.fiber.Engine$.$anonfun$create$1(AsyncCtrl.scala:147)
[error]         at spinal.core.fiber.AsyncThread.$anonfun$jvmThread$1(AsyncThread.scala:59)
[error]         at spinal.core.fiber.EngineContext.$anonfun$newJvmThread$1(AsyncCtrl.scala:39)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 18 s, completed 2023-3-7 15:47:15
`

Below are the error information:
`info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[info] For instance :
[info]   val result = Bool()
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[error] Exception in thread "main" java.lang.NullPointerException
[info]   val a,b = Bool()
[info]           
[error]         at spinal.core.sim.package$.setLong(package.scala:143)
[error]         at spinal.core.sim.package$SimBoolPimper.$hash$eq(package.scala:329)
[error]         at spinal.lib.sim.StreamDriver.<init>(Stream.scala:94)
[error]         at spinal.lib.sim.StreamDriver$.apply(Stream.scala:57)
[error]         at spinal.lib.bus.amba4.axilite.sim.AxiLite4Driver.<init>(AxiLite4Driver.scala:41)
[error]         at datapath.DataPathTb.<init>(DataPathSim.scala:9)
[error]         at datapath.DataPathSim$.$anonfun$main$1(DataPathSim.scala:67)
[error]         at spinal.core.internals.PhaseCreateComponent.$anonfun$impl$262(Phase.scala:2435)
[error]         at spinal.core.fiber.Engine$.$anonfun$create$1(AsyncCtrl.scala:147)
[error]         at spinal.core.fiber.AsyncThread.$anonfun$jvmThread$1(AsyncThread.scala:59)
[error]         at spinal.core.fiber.EngineContext.$anonfun$newJvmThread$1(AsyncCtrl.scala:39)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 18 s, completed 2023-3-7 15:47:15
`

aureoleday, 2023-03-07:
I moved the val aliteDrv = AxiLite4Driver(io.s_axi,clockDomain) inside the init() function and it's ok
Wonder why it work in the previous versions

Readon, 2023-03-07:
Personally, I don't think it's a good idea to mix the sim code with design code.
You used "io.s_axi" directly for AxiLite4Driver initial, so I guess you are defining the driver in a Component which is a design code.
However, the AxiLite4Driver is a sim code though.