SpinalHDL/SpinalHDL #1058: Implementing zero inserter to Stream

DEFAULTTEXHAPb, 2023-03-15:
I'm trying to implement simple upsampler, which inserts valid zeros between input valid stream samples. To complete this task constant always valid zeroStream val was created. Payload value of zeroStream is assigned to zero and valid value is assigned to True. But on elaboration phase SpinalHDL throws NO DRIVER ON error. Is there any way to implement constant stream in SpinalHDL?

```scala
package dsplib.experimental

import spinal.core._
import spinal.lib._
import spinal.lib.sim._
import spinal.lib.fsm._
import spinal.core.sim._

object Config {
  def spinal = SpinalConfig(
    targetDirectory = "hw/gen",
    defaultConfigForClockDomains = ClockDomainConfig(
      resetActiveLevel = HIGH
    ),
    onlyStdLogicVectorAtTopLevelIo = true
  )

  def sim = SimConfig.withConfig(spinal).withFstWave
}

class SimpleResampler[T <: Data](dataType: HardType[T], resamplingRate: Int, resampleType: String) extends Area {

  val resampleCounter: Counter = Counter(0 to resamplingRate-1) init 0

  val sourceStream: Stream[T] = Stream(dataType())
  val sinkStream: Stream[T] = Stream(dataType())
  val zeroStream: Stream[T] = Stream(dataType())
  val vecStream: Vec[Stream[T]] = Vec(sourceStream, zeroStream)

  
  // val zeroVal: Bits = Reg(Bits(dataType.getBitsWidth bits)) init 0 allowUnsetRegToAvoidLatch

  // zeroVal.asBits := 0
  
  when(sourceStream.fire) {
    resampleCounter.increment
  }

  // zeroStream.payload.asBits := zeroVal
  zeroStream.payload.asBits := 0
  zeroStream.valid := True

  sinkStream <-< StreamMux((resampleCounter.value === 0).asUInt, vecStream)

  def apply(that: Stream[T]): Stream[T] = {
    this.sourceStream << that
    this.sinkStream
  }

}

object SimpleResampler {
  def apply[T <: Data](dataType: HardType[T], resamplingRate: Int, resampleType: String): SimpleResampler[T] = {
    new SimpleResampler(dataType, resamplingRate, resampleType)
  }
}

case class Upsampler() extends Component {
  val io = new Bundle {
    val source = slave Stream(SInt(8 bits))
    val sink = master Stream(SInt(8 bits))
  }

  val ups = SimpleResampler(SInt(8 bits), 3, "UP")

  io.sink << ups(io.source)
}

object GenerateRtl extends App {
  Config.spinal.generateVerilog(
    Upsampler()
  ).printPruned()

  // Config.sim.compile(Upsampler()).doSim { dut =>
  //   dut.clockDomain.forkStimulus(10)
  //   StreamReadyRandomizer(dut.io.sink, dut.clockDomain)

  //   var idx: Int = 10
  //   StreamDriver(dut.io.source, dut.clockDomain) { payload =>
  //     idx -= 1
  //     payload.randomize
  //     idx >= 1
  //   }

  //   while (idx >= 1) {
  //     dut.clockDomain.waitSampling()
  //   }
  // }
}
```
The error log listed below
```
[Runtime] SpinalHDL v1.8.0    git head : 4e3563a282582b41f4eaafc503787757251d23ea
[Runtime] JVM max memory : 1982.0MiB
[Runtime] Current date : 2023.03.15 15:55:01
[Progress] at 0.000 : Elaborate components
[Progress] at 0.286 : Checks and transforms

**********************************************************************************************
[Warning] Elaboration failed (2 errors).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 0.393 : Elaborate components
[Progress] at 0.412 : Checks and transforms
Exception in thread "main" spinal.core.SpinalExit: 

 Error detected in phase PhaseCheck_noLatchNoOverride
********************************************************************************
********************************************************************************

NO DRIVER ON (toplevel/ups_zeroStream_payload :  SInt[8 bits]), defined at
    dsplib.experimental.Upsampler.$anonfun$ups$1(FixRounding.scala:66)
    dsplib.experimental.SimpleResampler.$anonfun$zeroStream$1(FixRounding.scala:28)
    spinal.lib.Stream.<init>(Stream.scala:56)
    spinal.lib.StreamFactory.apply(Stream.scala:35)
    spinal.lib.StreamFactory.apply(Stream.scala:40)
    dsplib.experimental.SimpleResampler.<init>(FixRounding.scala:28)
    dsplib.experimental.SimpleResampler$.apply(FixRounding.scala:56)
    dsplib.experimental.Upsampler.<init>(FixRounding.scala:66)
    dsplib.experimental.GenerateRtl$.$anonfun$new$2(FixRounding.scala:73)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

        at spinal.core.SpinalExit$.apply(Misc.scala:440)
        at spinal.core.SpinalError$.apply(Misc.scala:495)
        at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:174)
        at spinal.core.internals.PhaseContext.doPhase(Phase.scala:190)
        at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12(Phase.scala:2738)
        at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12$adapted(Phase.scala:2736)
        at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
        at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
        at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$7(Phase.scala:2736)
        at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
        at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2672)
        at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2667)
        at spinal.core.Spinal$.apply(Spinal.scala:405)
        at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:173)
        at dsplib.experimental.GenerateRtl$.delayedEndpoint$dsplib$experimental$GenerateRtl$1(FixRounding.scala:73)
        at dsplib.experimental.GenerateRtl$delayedInit$body.apply(FixRounding.scala:71)
        at scala.Function0.apply$mcV$sp(Function0.scala:39)
        at scala.Function0.apply$mcV$sp$(Function0.scala:39)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
        at scala.App.$anonfun$main$1$adapted(App.scala:80)
        at scala.collection.immutable.List.foreach(List.scala:431)
        at scala.App.main(App.scala:80)
        at scala.App.main$(App.scala:78)
        at dsplib.experimental.GenerateRtl$.main(FixRounding.scala:71)
        at dsplib.experimental.GenerateRtl.main(FixRounding.scala)
```

DEFAULTTEXHAPb, 2023-03-15:
Stream transaction extender helped me to implement the zero insertion. I just have not found any mention of this module in the documentation. Thanks

closed 2023-03-15 by DEFAULTTEXHAPb

Readon, 2023-03-16:
> Stream transaction extender helped me to implement the zero insertion. I just have not found any mention of this module in the documentation. Thanks

Yes, we still need more man power to write the doc.
You're welcome to write one. :)