SpinalHDL/SpinalHDL #1060: The clock or reset signal defined by ClockDomain.external won't work with addAttribute function

cheerryio, 2023-03-22:
my code is as follows:
```
class testSpinal(
                TEST_CLOCKDOMAIN:ClockDomain
                ) extends Component {
    val io=new Bundle{
        val out1=out Bits(4 bits)
    }

    new ClockingArea(TEST_CLOCKDOMAIN){
        val my_out1=CounterFreeRun(16)
        io.out1:=my_out1.asBits
    }
}

object testSpinal {
    def main(args: Array[String]): Unit = {
        SpinalConfig(
            mode=Verilog,
            defaultConfigForClockDomains=ClockDomainConfig(resetKind=SYNC,resetActiveLevel=HIGH),
            targetDirectory="verilog/novastar/Test",
            inlineRom=true
        ).generate({
            val TEST_CLOCKDOMAIN:ClockDomain=ClockDomain.external("test")
           //----------------------------LOOK AT HERE---------------------------
            TEST_CLOCKDOMAIN.clock.addAttribute("myAttribute","myValue")
          //----------------------------LOOK AT HERE----------------------------
            val inst=new testSpinal(TEST_CLOCKDOMAIN)

            inst
        })
    }
}
```

`addAttribute`  won't work....
When generating verilog, spinalHDL seems to just walk through the tags inside the top component to generate attributes. But I think signals defined by `ClockDomain.external` is really supposed to be able to addAttribute.

andreasWallner, 2023-03-23:
What's happening here is that you are not modifying the clock signal in your top component, but the a signal that has no real reference to a component - that's why SpinalHDL doesn't put your attribute anywhere.

You have to modify the clock signal local in the component where you want the attribute to show up.
That can be in the code of `testSpinal` directly:
```scala
...
  val io=new Bundle{
    val out1=out Bits(4 bits)
  }
  TEST_CLOCKDOMAIN.readClockWire.addAttribute("myAttribute","myValue")
  new ClockingArea(TEST_CLOCKDOMAIN){
...
```

If you want to keep it outside of the code of the component itself, you can use rework:
```scala
...
.generate({
      val TEST_CLOCKDOMAIN:ClockDomain=ClockDomain.external("test")
      val inst=new testSpinal(TEST_CLOCKDOMAIN)
      inst.rework {
        TEST_CLOCKDOMAIN.readClockWire.addAttribute("myAttribute","myValue")
      }
      inst
...
```

Also note that I used `readClockWire` instead of `clock` -> that returns you the real clock wire from inside the component.

cheerryio, 2023-03-24:
Thanks for your answer! This really has solved my problem!

closed 2023-03-24 by cheerryio