SpinalHDL/SpinalHDL #1061: How to achieve  A[B:12] A and B both are Reg     in spinal HDL

Mido-sun, 2023-03-22:
downto  can only accept Int para , But I want to use Reg

numero-744, 2023-03-22:
Reg is a value in a wire, so `A[B:12]` would have a size which changes during simulation, which is kind of equivalent to having your schematic changing during simulation :thinking: 

Maybe you want something like `A(B, 12 bits)` which is IIRC a 12-bit signal, sliced from A starting from index B?

Dolu1990, 2023-03-23:
let's say you want to read a selected byte in a 32 bits word : 

```scala
    val sel = in UInt(2 bits)
    val data = in Bits(32 bits)
    val muxedA = out Bits(8 bits)
    val muxedB = out Bits(8 bits)
    
    muxedA := data.subdivideIn(8 bits).read(sel) //Byte select
    muxedB := data(sel << 3, 8 bits)                    //Byte select too, rely  on synthesis to do a good job
```

```verilog
  reg        [7:0]    _zz_muxedA;
  wire       [4:0]    _zz_muxedB;

  assign _zz_muxedB = ({3'd0,sel} <<< 3);
  always @(*) begin
    case(sel)
      2'b00 : _zz_muxedA = data[7 : 0];
      2'b01 : _zz_muxedA = data[15 : 8];
      2'b10 : _zz_muxedA = data[23 : 16];
      default : _zz_muxedA = data[31 : 24];
    endcase
  end

  assign muxedA = _zz_muxedA;
  assign muxedB = data[_zz_muxedB +: 8];
```

Mido-sun, 2023-03-25:
I think the wrong way . @numero-744  you are right that it is wrong for using index like that

Mido-sun, 2023-03-25:
@Dolu1990  aha  that is not I want to imply

Dolu1990, 2023-03-25:
Ahhh i see XD

closed 2023-04-03 by Mido-sun