SpinalHDL/SpinalHDL #1063: Why there is no AXI4toapb3bridge lib

Amegodvivado, 2023-03-24:
Why there is no AXI4toapb3bridge lib but use AXI4sharedBridge, do we need to convert AXI4TOAxi4Share ourselves?

Amegodvivado, 2023-03-24:
HOW to convert？

Dolu1990, 2023-03-24:
Hi,

Using https://github.com/SpinalHDL/SpinalHDL/blob/6fba447046e0a846774a72b53676c5b318f9344f/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4.scala#L207
You can do myAxi4Bus.toShared()  and then connect this to the AXI4sharedBridge

Amegodvivado, 2023-03-24:
Thank you for your answer，I'm a newbie

closed 2023-03-24 by Amegodvivado

Dolu1990, 2023-03-27:
Yes ^^
Looks right to me :)

Amegodvivado, 2023-03-27:
Thank you very much for your reply in your busy schedule. By the way, how to implement the Axi4ToAhblite3 Bridge (or other bus convert )? Currently, there is no lib in Spinal, no def in AXI4. Haha~

Dolu1990, 2023-03-27:
Hmm i think currently there is nothing to bridge from AXI4 to AHBlite3
I think the only bridge for Axi4Toxxxx is to Apb3.

Amegodvivado, 2023-03-27:
> Hmm i think currently there is nothing to bridge from AXI4 to AHBlite3
> I think the only bridge for Axi4Toxxxx is to Apb3.

Ok,thank you again for your answer