SpinalHDL/SpinalHDL #1066: Any plans to implement LLVM circt?

panxxhub, 2023-03-27:


Readon, 2023-03-27:
The current state is that we are lacking of contributor, so if you're interested in this function the contribution is welcome.

If you have any question, the community is here to help.

Dolu1990, 2023-03-27:
One question is, what's about SpinalHDL -> Verilog -> LLVM circt ?
Is it realy bad ?
(as a way to save dev time and avoid a new backend ^^)

chenbo-again, 2023-03-30:
riscv: x0 ~ x31
llvm ir: x0 ~ x99999999...
it's useful for compiler frontend implementation and backend optimization

PeterWang-dev, 2025-03-02:
Is there any plan to involve LLVM circt in Spinal HDL now?

Dolu1990, 2025-03-02:
Hi,
There is no plan for it so far.

PeterWang-dev, 2025-03-03:
> Hi, There is no plan for it so far.

Thanks for your quick reply. The reason I ask this is Circt and LLVM infra may produce a further optimization of generated verilog. So I wonder if Spinal HDL have this backend.

Compared with Chisel, Spinal HDL is more formal and has a lot of tools simplifying the IP design (such as ClockDomain, Counter even ApbInterface). The simulation and verification are also great, especially for muti-thread and multi-clock. So it would be so cool if circt and more optimization from LLVM can be involved.

Dolu1990, 2025-03-03:
How good would be the SpinalHDL -> Verilog -> Circt flow ? Not optimal for sure, but would it work ? Or a lot of optimization would not be possible ?
(As much as possible, i'm trying to avoid having additional backends / maintenance)

PeterWang-dev, 2025-04-21:
> How good would be the SpinalHDL -> Verilog -> Circt flow ? Not optimal for sure, but would it work ? Or a lot of optimization would not be possible ? (As much as possible, i'm trying to avoid having additional backends / maintenance)

Sorry for the late reply. AFAIK, there is no other tools to do that kind of things.

PeterWang-dev, 2025-04-21:
> riscv: x0 ~ x31 llvm ir: x0 ~ x99999999... it's useful for compiler frontend implementation and backend optimization

What you say is IR for compilers not for circuit. LLVM CIRCT is another intermediate representation dedicated to digital logic.