SpinalHDL/SpinalHDL #1067: Unable to use Scala reflection in SpinalHDL

ZhaokunHu, 2023-03-28:
I'm trying to use Scala reflection in SpinalHDL, but I'm running into issues. It seems that SpinalHDL doesn't support reflection, which is causing problems for my project. Specifically, I need to be able to use reflection to access the methods in my SpinalHDL code.Here is my code：

`import spinal.core._
import spinal.lib._
import scala.reflect.runtime.{universe => ru}

case class RGB(channelWidth: Int) extends Bundle {
  val red = UInt(channelWidth bit)
  val green = UInt(channelWidth bit)
  val blue = UInt(channelWidth bit)
  def isBlack: Bool = red === 0 && green === 0 && blue === 0
}

class Example  extends Component {
  val source = slave Stream (RGB(8))
  val sink = master Stream (RGB(8))
  sink <-< source.throwWhen(source.payload.isBlack)
}

object ReflectionExample {
  def main(args: Array[String]): Unit = {
    val classSymbol = ru.typeOf[Example].typeSymbol.asClass
    val methods = classSymbol.toType.members.filter(_.isMethod).map(_.asMethod)
    methods.foreach { m =>
      println(s"Method name: ${m.name}, parameters: ${m.paramLists}")
    }
  }
}
`
I would appreciate it if someone could provide some guidance on how to use reflection. If there are any alternatives to reflection that would achieve the same results, I would be happy to explore those as well.

andreasWallner, 2023-03-29:
I'm not really deep into reflection, so I can't tell you where the difference really lies, but this works: 
```scala
object ReflectionExample {
  def main(args: Array[String]): Unit = {
    val classSymbol = classOf[Example]
    val methods = classSymbol.getDeclaredMethods
    methods.foreach { m =>
      println(s"Method name: ${m.getName}, parameters: ${m.getParameters}")
    }
  }
}
```
Does that help?
(I cheated a bit by looking at the places in SpinalHDL that use reflection)

If you want to know about possible alternatives we'd have to know what you want to achieve in the end - otherwise it's hard to know how you could get there w/o reflection...

ZhaokunHu, 2023-03-30:
Thank you for your response, and I apologize for not stating my question clearly.
I would like to obtain information about all the functions used in a SpinalHDL Component, similar to how the Example class uses the throwWhen function. Additionally, if any of these functions are used, I would like to know the input and output signals of the functions. Is it possible to achieve this using SpinalHDL or other methods?

Dolu1990, 2023-03-30:
Hi,

The endgoal being to get some documentation ?

andreasWallner, 2023-03-30:
if you really want a list of function calls then I'm not sure that you could do that for normal Scala code... As far as I know that information would be contained in the AST, but you can only get that for code that you can call reify on? 
The snippet that you posted above should also only print as member functions, not function calls in a method.

As Dolu1990 asked: what would you like to achieve in the end?

ZhaokunHu, 2023-03-30:
My ultimate goal is to attempt to implement the idea of #635 and generate images automatically using code. To obtain the information needed for drawing, I utilized the DataAnalyzer and ModuleAnalyzer functions to acquire the necessary data, and then attempted to plot all the information. When the logical structure of the code is relatively simple, the drawing effect is good. 
such as:
`import spinal.core._
import spinal.lib._
class MYSub0(cd: ClockDomain) extends Component {
  val io = new Bundle{
    val ai = in UInt(8 bits)
    val ao = out UInt(8 bits)
  }
  io.ao := RegNext(io.ai) init(0)
}
class MYSub1(cd: ClockDomain) extends Component {
  val io = new Bundle{
    val ai = in UInt(8 bits)
    val ao = out UInt(8 bits)
    val a2 = out UInt(8 bits)
  }
  io.ao := RegNext(io.ai) init(0)
  val cd2 = ClockDomain.external("adc")
  //alow another clockDomain not confict to default clockdomain
  val area = new ClockingArea(cd2){
    val tmp = RegNext(io.ai) init(0)
    val tmp2 = tmp + (RegNext(io.ai) init(0))
  }
  io.a2 := area.tmp2
}
class TopExample1 extends Component {
  val io = new Bundle{
    val a = in UInt(8 bits)
    val b0 = out UInt(8 bits)
    val b1 = out UInt(8 bits)
    val b2 = out UInt(8 bits)
  }
  val cd0 = ClockDomain.external("cp")
  val cd1 = ClockDomain.external("ap")
  val u_sub0 = cd0(new MYSub0(cd0)) // set u_sub0's default clockDomain with cd0
  val u_sub1 = cd1(new MYSub1(cd1)) // it allow anoter clockDomain in ther module
  u_sub0.io.ai := io.a
  u_sub1.io.ai := io.a
  io.b0 := u_sub0.io.ao
  io.b1 := u_sub1.io.ao
  io.b2 := u_sub1.io.a2
  val tmp = RegNext(io.a) init(0)
}
`
This code will automatically generate:
[image]
However, if the project utilizes many highly integrated functions or has complex logical structures (such as throwwhen, StateMachine, etc.), the system will automatically generate many registers after compilation to implement the functions. As I plot all the information, the resulting image becomes very complex and ugly. 
such as
`import spinal.core._
import spinal.lib._
case class RGB(channelWidth: Int) extends Bundle {
  val red = UInt(channelWidth bit)
  val green = UInt(channelWidth bit)
  val blue = UInt(channelWidth bit)
  def isBlack: Bool = red === 0 && green === 0 && blue === 0
}
class TopExample2  extends Component {
  val source = slave Stream (RGB(8))
  val sink = master Stream (RGB(8))
  sink <-< source.throwWhen(source.payload.isBlack)
}`
This code will automatically generate:
[image]
This code becomes very complex and ugly solely because of the use of the "throwwhen" function.
I am currently searching for a way to simplify these automatically generated registers. One solution I have considered is using the function names themselves to represent these registers for plotting, but it seems that this cannot be achieved technically. If there are any other methods to simplify these automatically generated registers, I am willing to try them out.

Dolu1990, 2023-03-31:
Hooo, then i think using reflection isn't the way, but instead using the internal datamodel of SpinalHDL should be enough. There is some doc about it there : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html

Then remain the case of for instance FSM detection. Here a way to detect that is to use the refOwner field of signal, it would point to the StateMachine instance and so on :)

andreasWallner, 2023-04-01:
Looks interesting, what are you using to generate the graphs?
If you want to look at the datamodel you could have a look here: https://github.com/andreasWallner/spinalStuff/blob/5dab5ea88b968d30abbc1c87929907925133436b/src/main/scala/andreasWallner/diagram/Diagrammer.scala#L112 - I have been on-and-off playing around with generating graphviz visualizations with the initial goal of "just" creating the same graphics as I would get from verilog - and then clean them up...
Note though on the code: I'm currently only extracting muxes, not other dependent assignments (when/switch), but it might be a place to start from.

ZhaokunHu, 2023-04-01:
Thank you both for your help. I will carefully try the suggestions you have given. 
I am using an open-source library from GitHub to generate the graphs:  https://github.com/davidthings/hdelk  @andreasWallner

closed 2023-04-03 by ZhaokunHu