SpinalHDL/SpinalHDL #1068: promote BusSlaveFactory

chenbo-again, 2023-03-30:
```scala
def onWrite(address: BigInt, documentation: String = null)(doThat: => Unit): Unit = {
    onWritePrimitive(
      address       = SingleMapping(address),
      haltSensitive = true,
      documentation = documentation
    )(doThat)
  }
```  
it's useful to add writedata param to doThat
it will be like:

```scala
def onWrite(address: BigInt, documentation: String = null)(doThat: Bits => Unit): Unit = {
    onWritePrimitive(
      address       = SingleMapping(address),
      haltSensitive = true,
      documentation = documentation
    )(doThat)
  }
```

Dolu1990, 2023-03-31:
Hi, 
The idea is more to trigger something. 
Currently, the idea (not very clean) was to use the nonStopWrite primitive to  get the bits to write.
I'm worrying that adding those Bits to onWrite will break backward compatibility. Maybe could "onWriteBits"  could be added instead ?

chenbo-again, 2023-04-09:
#1081

Readon, 2023-05-11:
The function onWrite is definied as register function whose logic would be activated while "a write transaction occur on address".
In many buses that data and address are transferred almost independently, so data might not available when transaction start with  just address.
So use nonStopWrite might be a better way to avoid misunderstand.