SpinalHDL/SpinalHDL #1070: Confusion when using history

zwxb, 2023-03-31:
### Discussed in https://github.com/SpinalHDL/SpinalHDL/discussions/1069

<div type='discussions-op-text'>

<sup>Originally posted by **zwxb** March 30, 2023</sup>
### When using History,the following source code:
```
case class SerilIIRV2(Qi: Int) extends Component {
  val io = new Bundle {
    val input = slave(Flow(SInt(Qi bits)))
    val output = out(SInt(16 bits))
  }
  noIoPrefix()
  val YBuffer = History((io.input.payload), 4, DValid, S(0, 16 bits))
  val DValid = Reg(Bool()) init (False)
  DValid := io.input.valid
  io.output := YBuffer(3)
}
```
### Generated verilog:
```
`timescale 1ns/1ps

module SerilIIRV2 (
  input               input_valid,
  input      [15:0]   input_payload,
  output     [15:0]   output_1,
  input               clk,
  input               reset
);

  wire       [15:0]   YBuffer_0;
  reg        [15:0]   YBuffer_1;
  reg        [15:0]   YBuffer_2;
  reg        [15:0]   YBuffer_3;
  reg                 DValid;

  assign YBuffer_0 = input_payload;
  assign output_1 = YBuffer_3;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      YBuffer_1 <= 16'h0;
      YBuffer_2 <= 16'h0;
      YBuffer_3 <= 16'h0;
      DValid <= 1'b0;
    end else begin
      YBuffer_1 <= YBuffer_0;
      YBuffer_2 <= YBuffer_1;
      YBuffer_3 <= YBuffer_2;
      DValid <= input_valid;
    end
  end

endmodule

```
### I just changed the Dvalid declaration position,the following source code:
```

case class SerilIIRV2(Qi: Int) extends Component {
  val io = new Bundle {
    val input = slave(Flow(SInt(Qi bits)))
    val output = out(SInt(16 bits))
  }
  noIoPrefix()
  val DValid = Reg(Bool()) init (False)
  val YBuffer = History((io.input.payload), 4, DValid, S(0, 16 bits))

  DValid := io.input.valid
  io.output := YBuffer(3)
}
```
### Generated verilog:
```
`timescale 1ns/1ps

module SerilIIRV2 (
  input               input_valid,
  input      [15:0]   input_payload,
  output     [15:0]   output_1,
  input               clk,
  input               reset
);

  reg                 DValid;
  wire       [15:0]   YBuffer_0;
  reg        [15:0]   YBuffer_1;
  reg        [15:0]   YBuffer_2;
  reg        [15:0]   YBuffer_3;

  assign YBuffer_0 = input_payload;
  assign output_1 = YBuffer_3;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      DValid <= 1'b0;
      YBuffer_1 <= 16'h0;
      YBuffer_2 <= 16'h0;
      YBuffer_3 <= 16'h0;
    end else begin
      if(DValid) begin
        YBuffer_1 <= YBuffer_0;
      end
      if(DValid) begin
        YBuffer_2 <= YBuffer_1;
      end
      if(DValid) begin
        YBuffer_3 <= YBuffer_2;
      end
      DValid <= input_valid;
    end
  end
endmodule
```
### I don't know why this is happening because the scala syntax isn't used correctly?</div>

Dolu1990, 2023-03-31:
Statements in Scala are executed in their definition order, so 
```scala
  val YBuffer = History((io.input.payload), 4, DValid, S(0, 16 bits))
  val DValid = Reg(Bool()) init (False)
  ``` 

History will see DValid as null, which happens to be used by the history tool to mean "always enabled"

zwxb, 2023-03-31:
Thank you for your reply, I understand the cause of the problem.

closed 2023-03-31 by zwxb