SpinalHDL/SpinalHDL #1071: BundleWithAssign trait

xueweiwujxw, 2023-03-31:
I always use `<<` or `>>` to connect the bus. However, I must to implement these functions in my custom bundles.

xueweiwujxw, 2023-03-31:
```scala
object BundleWithAssignTraitTest extends App {
  case class TestBundle()
      extends Bundle
      with IMasterSlave
      with BundleWithAssignTrait[TestBundle] {
    val valid = Bool
    val ready = Bool

    def asMaster(): Unit = {
      inWithNull(ready)
      outWithNull(valid)
    }

    def connectFrom(that: TestBundle): TestBundle = {
      that.ready := this.ready
      this.valid := that.valid
      that
    }
  }

  case class TestModule() extends Component {
    val io = new Bundle {
      val s = slave(TestBundle())
      val m = master(TestBundle())
    }

    val save = Reg(TestBundle())
    io.s >> save
    io.m << save
  }

  SpinalConfig().generateVerilog(TestModule())
}
```

Dolu1990, 2023-03-31:
Hi
That's a good idea.
Hmm about the BundleWithAssignTrait name, maybe more something like :
IConnectable ?

xueweiwujxw, 2023-03-31:
> Hi That's a good idea. Hmm about the BundleWithAssignTrait name, maybe more something like : IConnectable ?

It sounds great. I will push a new commit.

xueweiwujxw, 2023-03-31:
I forgot to create a new branch.
I closed the pr #1072 and opened a new pr #1073

PR #1073 (2023-10-16): Iconnectable

Closes #1071 

# Context, Motivation & Description

Add IConnectable trait to connect bus

Example:
```scala
object BundleWithAssignTraitTest extends App {
  case class TestBundle()
      extends Bundle
      with IMasterSlave
      with IConnectable[TestBundle] {
    val valid = Bool
    val ready = Bool

    def asMaster(): Unit = {
      inWithNull(ready)
      outWithNull(valid)
    }

    def connectFrom(that: TestBundle): TestBundle = {
      that.ready := this.ready
      this.valid := that.valid
      that
    }
  }

  case class TestModule() extends Component {
    val i [... cut]
closed 2023-10-16 by Dolu1990