SpinalHDL/SpinalHDL #1077: Feature request: `Vec` slicing

jonnykl, 2023-04-07:
Is there a reason why slicing of `Vec`s are not implemented yet?

I'm adding that functionality using an implicit class to `Vec`:

```scala
def apply(range: Range): Vec[T] = Vec(for (i <- range.low to range.high) yield vec(i))
```

Maybe this could be added so SpinalHDL nativly supports it?

Dolu1990, 2023-04-07:
ahhh right, that would be a nice one to add :)

jonnykl, 2023-04-07:
Note: Writing `range.low to range.high` instead of just `range` is important to keep the order correct when using `downto`.

jonnykl, 2023-04-07:
I also have some more methods you maybe want to add, here is what I'm currently using:

```scala
  /** Extended Vec that can be initialized using an init method.
   *
   *  Can be implicitly converted back to Vec by InitVecToVec.
   *
   *  @param vec the vector
   *  @param range specifies the elements which will be initialized by the init method
   */
  case class InitVec[T <: Data](vec: Vec[T], range: Range) {
    /** Initializes the previously specified elements.
     *
     *  @param data will be assigned to each of the specified elements
     *  @return underlying vector.
     */
    def init(data: T = null.asInstanceOf[T]) = {
      if (data != null) {
        for (i <- range) {
          vec(i) := data
        }
      }

      vec
    }
  }

  implicit class AdditionalVecOps[T <: Data](vec: Vec[T]) {
    /** Right-shifts the vector without changing its size.
     *
     *  @param n shift count
     *  @return initializable vector.
     */
    def |>>(n: Int) = {
      val newVec = cloneOf(vec)

      for (i <- 0 until vec.length-n) {
        newVec(i) := vec(i+n)
      }

      InitVec(newVec, vec.length-n until vec.length)
    }

    /** Right-shifts the vector by discarding the lower elements.
     *
     *  @param n shift count
     *  @return vector.
     */
    def >>(n: Int) = {
      takeHigh(Seq(vec.size-n, 0).max)
    }

    /** Left-shifts the vector without changing its size.
     *
     *  @param n shift count
     *  @return initializable vector.
     */
    def |<<(n: Int) = {
      val newVec = cloneOf(vec)

      for (i <- 0 until vec.length-n) {
        newVec(i+n) := vec(i)
      }

      InitVec(newVec, 0 until n)
    }

    /** Left-shifts the vector by creating the lower elements.
     *
     *  @param n shift count
     *  @return initializable vector.
     */
    def <<(n: Int) = {
      val newVec = Vec(vec.dataType(), vec.size+n)
      vec.zipWithIndex.foreach { case (element, i) => newVec(i+n) := element }
      InitVec(newVec, 0 until n)
    }

    /** Left-shifts the vector without changing its size.
     *
     *  @param n shift count
     *  @return initializable vector.
     */
    def |<<=(n: Int): Vec[T] = {
      for (i <- 0 until vec.length-n) {
        vec(i+n) := vec(i)
      }

      vec
    }

    /** Resizes the vector.
     *
     *  @param size new size
     *  @return initializable vector.
     */
    def resize(size: Int): InitVec[T] = {
      InitVec(Vec(for (i <- 0 until size) yield {
        if (i < vec.size) {
          vec(i)
        } else {
          vec.dataType()
        }
      }), vec.size until size)
    }

    /** Returns the lowest n elements.
     *
     *  @param n number of elements
     *  @return sliced vector.
     */
    def takeLow(n: Int): Vec[T] = this(0 until n)

    /** Returns all except the lowest n elements.
     *
     *  @param n number of elements
     *  @return sliced vector.
     */
    def dropLow(n: Int): Vec[T] = this(n until vec.length)

    /** Returns the highest n elements.
     *
     *  @param n number of elements
     *  @return sliced vector.
     */
    def takeHigh(n: Int): Vec[T] = dropLow(vec.length - n)

    /** Returns all except the highest n elements.
     *
     *  @param n number of elements
     *  @return sliced vector.
     */
    def dropHigh(n: Int): Vec[T] = takeLow(vec.length - n)

    /** Concatenates another vector.
     *
     *  @param other the other vector
     *  @return new vector.
     */
    def ###(other: Vec[T]): Vec[T] = Vec(other ++ vec)

    /** Returns a slice of the vector.
     *
     *  @param range slice range
     *  @return sliced vector.
     */
    def apply(range: Range): Vec[T] = Vec(for (i <- range.low to range.high) yield vec(i))

    /** Subdivides the vector in multiple vectors of the same length.
     *
     *  @param partLength length of one part
     *  @return vector of all parts.
     */
    def subdivideIn(partLength: Int): Vec[Vec[T]] = {
      assert(vec.length%partLength == 0, "length of the vec must be a multiple of part length")

      val count = vec.length/partLength
      Vec(for (i <- 0 until count) yield vec(i*partLength until (i+1)*partLength))
    }
  }

  /** Converts InitVec back to Vec. */
  implicit def InitVecToVec[T <: Data](initVec: InitVec[T]) = initVec.vec
```

I'm not 100% happy with my solution for the initialization when shifting, but it works. Example:

```scala
vec := vec |>> 2 init(B(0).resized)
```

Dolu1990, 2023-04-08:
So, InitVec is integrating too much things, instead : 
myVec(myRange) initEach(myLiteral)  //Same as myVec(myRange).foreach(_.init(myLiteral)

>> << |>> |<< why not ^^

\#\#\#  What's about  instead implementing the ++ function in the Vec class ?

subdivideIn(partLength: Int) may be subdivideIn(partLength: XXX)  , XXX being SlicesCount and ElementCount ? or something like it ?

Thanks :D

jonnykl, 2023-04-11:
> So, InitVec is integrating too much things, instead :
> myVec(myRange) initEach(myLiteral) //Same as myVec(myRange).foreach(_.init(myLiteral)

Looks good.

> ### What's about instead implementing the ++ function in the Vec class ?

I chose `###` because `##` is already used in SpinalHDL for bitvectors. Using `###` makes clear that it's a SpinalHDL operator. I think using `++` would lead to confusion because it looks like someone is doing something with some sort of (scala) lists.

> subdivideIn(partLength: Int) may be subdivideIn(partLength: XXX) , XXX being SlicesCount and ElementCount ? or something like it ?

I think it would make sense to reuse `SlicesCount` from `BitVector.subdivideIn`. Maybe the `strict` parameter could also be added to achieve exactly the same functionality. As I think it's useful to be able to specify the length of the slices (as I implemented it), there could also be added an corresponding overloaded method (for both, `Vec` and `BitVector`).

jonnykl, 2023-04-11:
> > So, InitVec is integrating too much things, instead :
> > myVec(myRange) initEach(myLiteral) //Same as myVec(myRange).foreach(_.init(myLiteral)
> 
> Looks good.

Ahh no, it's not doing the same. My `init` function only initializes the new elements when doing a shift whereas your proposal would initialize all elements of the new vector.

Dolu1990, 2023-04-14:
> I chose ### because ## is already used in SpinalHDL for bitvectors. Using ### makes clear that it's a SpinalHDL operator. I think using ++ would lead to confusion because it looks like someone is doing something with some sort of (scala) lists.

The thing is that Vec in SpinalHDL is a emulation layer to access multiple signal as an array. In other words, Vec is like a Scala collection with a few extra thing, it doesn't contains hardware signals but instead refer to hardware signals.

So depending what you want to do with Vec concatenation, both ++ and ### could have different meanings.

++ would mean to create a new Vec which refer to the existing signals, while ### would mean to create a new Vec which refer to copies of the existing signals.

> Maybe the strict parameter could also be added to achieve exactly the same functionality. As I think it's useful to be able to specify the length of the slices (as I implemented it), there could also be added an corresponding overloaded method (for both, Vec and BitVector).

Sure :)

> def init(data: T = null.asInstanceOf[T])

Ahhh i missreaded it, i were thinking it was to initialise registers, not to assign a value.
What is the intend of this function (as you wished) ?

jonnykl, 2023-04-14:
> The thing is that Vec in SpinalHDL is a emulation layer to access multiple signal as an array. In other words, Vec is like a Scala collection with a few extra thing, it doesn't contains hardware [...]

Ok, makes sense. Then I would suggest to implement both operators as you described it. It would be important to document the difference as that's not obvious.

> Ahhh i missreaded it, i were thinking it was to initialise registers, not to assign a value.
> What is the intend of this function (as you wished) ?

When doing a "normal" logical bit shift, the new bits are always filled with zeros as "neutral" bits. For `Vec`s you could also fill the new elements with zeros. My `init` function should assign the given data to all "new" elements (which would otherwise be uninitialized or initialized with zeros). Example:

```scala
val vec = Vec(B"01001001")

val shiftedVec = vec << 2
// shiftedVec(2) = B"01001001"
// shiftedVec(1) and shiftedVec(0) would be uninitialized here

val shiftedVec = vec << 2 init(B"11111111")
// shiftedVec(2) = B"01001001"
// shiftedVec(1) = B"11111111"
// shiftedVec(0) = B"11111111"
```

As already noted, I'm not really happy with that solution. At least the name `init` should be changed as it does no initialization but an assignment.

Dolu1990, 2023-04-17:
Ahhh i see, hmm then mayben instead of naming it InitVec, it could be part of AdditionalVecOps ? and named rangedAssign ? or something like it ?

jonnykl, 2023-04-24:
> it could be part of AdditionalVecOps ?

Sounds like you want to directly use my class. Why not integrate it directly into the `Vec` class? I just chose this way because it doesn't required a patched SpinalHDL version.

Besides that, if `InitVec` should be integrated into the `Vec` class, it somehow needs to keep track of the new/unassigned elements (that's what the `range` parameter of `InitVec` does).

> and named rangedAssign ? or something like it ?

What about `assignNew()` (instead of `init()` in my previous example code)? This name makes clear that it is an assignment and that it applies to the "new" elements.

Dolu1990, 2023-04-28:
>  I just chose this way because it doesn't required a patched SpinalHDL version.

Ahhh right :) I was taking it as a PR ^^

> assignNew

How does it know which element are new ?

Maybe the best to avoid being too specific on one usage would be to decompose things : 

myVec(x to y).foreach(_ := z) ?

Readon, 2025-06-17:
Does those operations all necessary?

PR #1745 (2025-06-17): add slice to Vec, which fix #1077

Closes #1077

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [x] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?
closed 2025-06-17 by Dolu1990