SpinalHDL/SpinalHDL #1082: What are the differences between lib/sdr and lib/xdr?

LurenAA, 2023-04-09:
1. What are the differences between lib/sdr and lib/xdr? Does lib/sdr only support SDRAM, while lib/xdr supports both SDRAM and DDRx SDRAM? 

2. Do lib/sdr and lib/xdr now only support a SDRAM chip, not a SDRAM memory module(such as MT16KTF1G64HZ - 8GB, a DDR3L SDRAM SODIMM)? 

I am a novice to SpinalHDL, I would be very grateful if you could help me. 😁

Readon, 2023-04-09:
I also traced a little bits.
The definition of sdr's IO is as below.
[image]

Here it is for xdr:
[image]

The most obvious difference is the ODT signal is introduced.
This signal is defined firstly while DDR2 standard.

Dolu1990, 2023-04-10:
lib/sdr is SDRAM gen 1 only, while lib/xdr could "potentialy" target any DRAM tech, as long as there is a phy implemented for it.

Currently both were only used for regular chip. Not modules.
I guess as long as you use modules with single rank it could work ?

andreasWallner, 2023-04-22:
@LurenAA Did that answer your question? If so, can we close this?

closed 2023-04-23 by LurenAA