SpinalHDL/SpinalHDL #1083: [Bug] BusSlaveFactory

chenbo-again, 2023-04-10:
```
val factory = AhbLite3SlaveFactory(io.ahblite)
val factory1 = Apb3SlaveFactory(io.apb)
val streamUnbuffered = factory.createAndDriveFlow(Bits(1 bits), address = 0x010).toStream
```
BusSlaveFactory.createAndDriveFlow is unusable when it's a ahblite bus, but it's usable when it's a apb bus.
```
toplevel/null isn't able generate the following requests :
[error] BusSlaveFactoryNonStopWrite((toplevel/??? :  Bits[1 bits]),0,null) at
[error]     spinal.lib.bus.amba3.ahblite.AhbLite3SlaveFactory.<init>(AhbLite3SlaveFactory.scala:37)
[error]     spinal.lib.bus.amba3.ahblite.AhbLite3SlaveFactory$.apply(AhbLite3SlaveFactory.scala:34)
```

andreasWallner, 2023-04-10:
I think that is due to `AhbLite3SlaveFactory` never generating the nonstop writes needed...
I did a quick check with `super.doNonStopWrite(bus.HWDATA)` added in the `build` method, from quick inspection this looks promising but I guess there should be a test for that.
Would you want to make a PR with a fix for that?

chenbo-again, 2023-04-10:
sure, I will do later.

chenbo-again, 2023-04-11:
I wonder how to trigger main func in `tester` folder, I can not do that, sorry for my poor knowledge of sbt.

andreasWallner, 2023-04-11:
Normally there should not be a need for a a main function.
We have two types of testcases, ones using spinalsim and others using cocotb. But both build on scalatest, so you run the tests via `test` or `testOnly name.of.the.test.class`.

Example for:
- spinalsim testcase: https://github.com/SpinalHDL/SpinalHDL/blob/dev/tester/src/test/scala/spinal/tester/scalatest/Axi4ToAxiLite4Tester.scala#L27
- cocotb testcase: https://github.com/SpinalHDL/SpinalHDL/blob/dev/tester/src/test/scala/spinal/tester/scalatest/Axi4SlaveFactoryTester.scala#L34

(`SpinalAnyFunSuite` and `SpinalTesterCocotbBase` both extend scalatest `AnyFunSuite`s)