SpinalHDL/SpinalHDL #1089: AutoConnect / <> bug?

dokleina, 2023-04-18:
I'm not sure what's wrong with this code, but I've narrowed it down to the "<>" calls within `UartCtrl`.

Basically, I'm trying to drive Uart through what should be a black boxed Xilinx IP's generated clock. The Uart code I borrowed from `UartCtrlUsageExample` and wrapped in a clock domain.

```scala

import spinal.core._
import spinal.lib._
import spinal.lib.com.uart._

// Just a pass through for what should be a blackbox of Xilinx's Clock Wizard
class pll_dummy extends Component {
  val io = new Bundle {
    val clk_in1 = in(Bool())
    val clk_out1 = out(Bool())
    val locked = out(Bool())
    val reset = in(Bool())
  }

  io.clk_out1 := io.clk_in1
  io.locked := !io.reset
}

class MyTopLevel extends Component {

  // Top level IO
  val io = new Bundle {
    // UART IO
    val uart     = master(Uart())

    // Discrete IO
    val switches = in  Bits(8 bits)
    val leds     = out Bits(8 bits)
  }

  val pll = new pll_dummy()

  val pllDomain = ClockDomain(
    pll.io.clk_out1,
    !pll.io.locked,
    frequency = FixedFrequency(100 MHz)
  )

  val pllArea = new ClockingArea(pllDomain) {
    val uartCtrl: UartCtrl = UartCtrl(
      config = UartCtrlInitConfig(
        baudrate = 921600,
        dataLength = 7, // 8 bits
        parity = UartParityType.NONE,
        stop = UartStopType.ONE
      )
    )
    uartCtrl.io.uart <> io.uart

    io.leds := uartCtrl.io.read.toFlow.toReg()

    val write = Stream(Bits(8 bits))
    write.valid := CounterFreeRun(2000).willOverflow
    write.payload := io.switches
    write >-> uartCtrl.io.write
  }
}

object MyTopLevel {
  def main(args: Array[String]) {
    SpinalConfig(
      mode = VHDL,
      defaultClockDomainFrequency = FixedFrequency(125 MHz)
    ).generate(new MyTopLevel)
  }
}
```

This code compiles if you comment out:
```scala
val pll = new pll_dummy()
```
and update the `pllDomain`'s clock and reset with something valid, for example:
```scala
val pllDomain = ClockDomain(
  ClockDomain.current.readClockWire,
  ClockDomain.current.readResetWire,
  frequency = FixedFrequency(100 MHz)
)
```

The errors I'm seeing are:
```
Exception in thread "main" spinal.core.SpinalExit: 

 Error detected in phase PhaseCreateComponent
********************************************************************************
********************************************************************************

HIERARCHY VIOLATION, (toplevel/??? : out Bool) can't be used in null at
    MyTopLevel$$anon$3.<init>(ZyboTopLevel.scala:49)
    MyTopLevel.<init>(ZyboTopLevel.scala:39)
    MyTopLevel$$anonfun$main$1.apply(ZyboTopLevel.scala:65)
    MyTopLevel$$anonfun$main$1.apply(ZyboTopLevel.scala:65)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

HIERARCHY VIOLATION, (toplevel/??? : in Bool) can't be used in null at
    MyTopLevel$$anon$3.<init>(ZyboTopLevel.scala:49)
    MyTopLevel.<init>(ZyboTopLevel.scala:39)
    MyTopLevel$$anonfun$main$1.apply(ZyboTopLevel.scala:65)
    MyTopLevel$$anonfun$main$1.apply(ZyboTopLevel.scala:65)
    spinal.sim.JvmThread.run(SimManager.scala:51)
```

I have confirmed that the "null" referenced in the error is merely a missing name. I've debugged `UartCtrlUsageExample` and confirmed the same name is given to it's `uartCtrl` instance. I don't know how to fix it, but it appears to be benign and misleading.

The text of the HIERARCHY VIOLATION seems to be coming from `autoConnectBaseImpl` in Data. I've found some "<>" usage in `UartCtrl` which might be part of the problem (maybe?).

Is this actually a hierarchy problem? Am I using `ClockingArea` wrong?
I'm lost as to why simply including the additional Component breaks everything.

Any help would be appreciated!

Dolu1990, 2023-04-18:
Hi,

I tried the code snipet,  but can't reproduce the bug, which version of SpinalHDL are you using ?

dokleina, 2023-04-18:
I'm currently building this project out of a personal fork at https://github.com/dokleina/SpinalHDL/commit/78443105c5ed314bf76dd87f51aedf2a94292e16

I've managed to get sbt to build the fork cloned into a submodule under ./ext/SpinalHDL with this build.sbt:
```scala
ThisBuild / name := "TestProject"
ThisBuild / version := "0.1"
ThisBuild / scalaVersion := "2.11.12"

autoCompilerPlugins := true

lazy val spinalHdlCore = ProjectRef(file("./ext/SpinalHDL/"), "core")
lazy val spinalHdlLib = ProjectRef(file("./ext/SpinalHDL/"), "lib")
lazy val spinalIdslPlugin = ProjectRef(file("./ext/SpinalHDL/"), "idslplugin")

lazy val testProject = (project in file("."))
  .aggregate(spinalHdlCore, spinalHdlLib, spinalIdslPlugin)
  .dependsOn(spinalHdlCore, spinalHdlLib, spinalIdslPlugin)

fork := true
```

Dolu1990, 2023-04-18:
hmm one thing is if i take your snipet of code, the pll inputs are unconnected ?

dokleina, 2023-04-18:
I've found it doesn't matter if I connect them through a BlackBox to the default clock domain or not. The mere instantiation of the PLL component causes the crash.

dokleina, 2023-04-19:
I gave up on trying to have my project rebuild Spinal as part of the top level project. I've switched to publishing the dev branch build locally and pointing to that.

It works now.

Closing this because I'm just bad at sbt 🤷

closed 2023-04-19 by dokleina