SpinalHDL/SpinalHDL #1090: Encountering Issues when Obtaining ClockDomains

ZhaokunHu, 2023-04-18:
Here is my code:
```scala

import spinal.core._
import spinal.lib.tools.ModuleAnalyzer

class MYSub0(cd: ClockDomain) extends Component {
  val io = new Bundle {
    val ai = in UInt (8 bits)
    val ao = out UInt (8 bits)
  }
  io.ao := RegNext(io.ai) init (0)
}

class MYSub1(cd: ClockDomain) extends Component {
  val io = new Bundle {
    val ai = in UInt (8 bits)
    val ao = out UInt (8 bits)
    val a2 = out UInt (8 bits)
  }

  io.ao := RegNext(io.ai) init (0)
  val cd2 = ClockDomain.external("adc")
  val area = new ClockingArea(cd2) {
    val tmp = RegNext(io.ai) init (0)
    val tmp2 = tmp + (RegNext(io.ai) init (0))
  }
  io.a2 := area.tmp2
}

class Example1 extends Component {
  val io = new Bundle {
    val a = in UInt (8 bits)
    val b0 = out UInt (8 bits)
    val b1 = out UInt (8 bits)
    val b2 = out UInt (8 bits)
  }
  val cd0 = ClockDomain.external("cp")
  val cd1 = ClockDomain.external("ap")

  val u_sub0 = cd0(new MYSub0(cd0)) 
  val u_sub1 = cd1(new MYSub1(cd1)) 

  u_sub0.io.ai := io.a
  u_sub1.io.ai := io.a

  io.b0 := u_sub0.io.ao
  io.b1 := u_sub1.io.ao
  io.b2 := u_sub1.io.a2

  val tmp = RegNext(io.a) init (0)
}

class Example2 extends Component {
  val io = new Bundle {
    val i = in port Bool()
    val subo = out port Bool()
    val subo2 = out port Bool()
  }
  val sub = new Component {
    val i = in port Bool()
    val o = out port Bool()

    val myClockDomain = ClockDomain.external("clk")
    val coreArea = new ClockingArea(myClockDomain) {
      o := RegNext(i)
    }
  }
  val sub2 = new Component {
    val i = in port Bool()
    val o = out port Bool()

    val myClockDomain = ClockDomain.external("clk")
    val coreArea = new ClockingArea(myClockDomain) {
      o := RegNext(i)
    }
  }
  sub.i := io.i
  io.subo := sub.o
  sub2.i := io.i
  io.subo2 := sub2.o
}

object ClockExample {
  def searchClockDomains(rtl: SpinalReport[Component]): Unit = {
    val module = rtl.toplevel
    val moduleAna = new ModuleAnalyzer(module)
    val clockSet = moduleAna.getClocks(_ => true)
    println("clocks:" + clockSet)
    val inputs = moduleAna.getInputs
    for (input <- inputs) println(s"ClockDomain of $input is ${input.clockDomain}")
  }

  def main(args: Array[String]): Unit = {
    searchClockDomains(SpinalVerilog(new Example1))
    searchClockDomains(SpinalVerilog(new Example2))
  }
}
```
The outputs  of Example1 are as follows:
```
clocks:Set(clk, cp_clk, ap_clk, adc_clk)
ClockDomain of (toplevel/io_a : in UInt[8 bits]) is clk
ClockDomain of (toplevel/clk : in Bool) is clk
ClockDomain of (toplevel/reset : in Bool) is clk
ClockDomain of (toplevel/cp_clk : in Bool) is cp_clk
ClockDomain of (toplevel/cp_reset : in Bool) is cp_clk
ClockDomain of (toplevel/ap_clk : in Bool) is ap_clk
ClockDomain of (toplevel/ap_reset : in Bool) is ap_clk
ClockDomain of (toplevel/adc_clk : in Bool) is ap_clk
ClockDomain of (toplevel/adc_reset : in Bool) is ap_clk
```
Why the ClockDomain of  (toplevel/adc_clk : in Bool) is ap_clk?

The outputs of Example2 are as follows:
```
clocks:Set(clk_clk, clk_clk)
ClockDomain of (toplevel/io_i : in Bool) is clk
ClockDomain of (toplevel/clk_clk : in Bool) is clk
ClockDomain of (toplevel/clk_reset : in Bool) is clk
ClockDomain of (toplevel/clk_clk_1 : in Bool) is clk
ClockDomain of (toplevel/clk_reset_1 : in Bool) is clk
```
1. Why does the clockSet include two identical ClockDomains instead of (clk_clk, clk_clk_1)? How can I differentiate between these two ClockDomains?
2. Why is the ClockDomain of all inputs is 'clk',The clockSet even don't have 'clk' ?

Dolu1990, 2023-04-19:
Hi,

The way how ClockDomain.external(name) works, is that it add a new external clockdomain where the clock is named name_clk. The ClockDomain instance itself will be named as 'name'.
So you realy need to call ClockDomain.external(name) with an unique name each time. You may want to give the name as argument of the sub component.

More generaly, what would be your expectations ?

andreasWallner, 2023-04-20:
As mentioned in my review comment:
Why do you use the name (`getName`) to distinguish between the clocks?
Using the object from the `.clock` property itself (https://github.com/SpinalHDL/SpinalHDL/blob/8acff23015d4107e4a915f53d0794deade884f08/core/src/main/scala/spinal/core/ClockDomain.scala#L279) instead of a name should allow you to distinguish between between the clock domains. (you'll still not correctly distinguish between a derived clock domain with an enable signal and it's parent clock domain)

ZhaokunHu, 2023-04-20:
> As mentioned in my review comment: Why do you use the name (`getName`) to distinguish between the clocks? Using the object from the `.clock` property itself ( [...]

I am using `getName` to distinguish clock domains because, as shown in `Example1`, the only way to obtain the clock domain of a signal is by using the `.ClockDomain` function. However, for signals like `ClockDomain of (toplevel/adc_clk : in Bool) is ap_clk`, I know that `ap_clk` is the parent clock domain of `adc_clk`, but when generating diagrams, the `adc_clk` port using `ap_clk`'s color can be confusing. To avoid this confusion, I need to use `getName` to distinguish the clock domains and assign colors accordingly.

Regarding my issue in `Example2`, you can see that the print `ClockDomain of (toplevel/clk_reset_1 : in Bool) is clk`. First, HDElk does not support using the same name in one node, so I had to name the second `clk_clk` as `clk_clk_1` in HDElk, just like the signal name. However, you can see from the output that the clock domain I obtained (`clk`) does not match the signal name (`clk_clk_1`)( In my program, I am using `getClock` to initialize `clkmap`, but I am unable to assign a color to `clk` because I cannot obtain its information). Although I can distinguish between the clock domains by using the ClockDomain object itself, it cannot match the signal(I can only obtain the information from the object that this clock is named `clk_clk`). Therefore, my question is how to match the `ClockDomain` with the signal in the needed way(What color should be  given to a signal ).

In addition, I think this bug is difficult to handle in the current framework. I think suggest  users avoid defining two clock domains with the same name is better...

andreasWallner, 2023-04-20:
What I was trying to say: instead of the names you can (have to...) use the ClockDomain objects directly (or the `clock` and `reset` properties of them). Those are objects that you can compare and/or use as keys in a map, so there is no need to try to use the names (which might not be unique as you demonstrated). This is not a bug since names are claimed to be unique.
This could work in your case similar to this: 
```scala
object TestClocks extends App {
  class Inner extends Component {
    val i = in port Bool()
    val o = out port Bool()
    val cd = new ClockingArea(ClockDomain.external("clk")) {
      val registered = RegNext(i)
      o := registered
    }
  }
  val report = SpinalVerilog(new Component {
    val i = in port Bits(3 bit)
    val o = out port Bits(3 bit)
    val inner0 = new Inner
    val inner1 = new Inner
    inner0.i := i(0)
    o(0) := inner0.o
    inner1.i := i(1)
    o(1) := inner1.o

    o(2) := RegNext(i(2))
  })

  val analyzer = new ModuleAnalyzer(report.toplevel)
  val clockDomains = analyzer.getClocks.toList
  println(clockDomains mkString(" "))

  val colors = Map(clockDomains(1) -> 1, clockDomains(2) -> 2)

  assert(clockDomains(0) != clockDomains(1))
  assert(clockDomains(1) == report.toplevel.inner0.cd.registered.clockDomain)
  assert(clockDomains(2) == report.toplevel.inner1.cd.registered.clockDomain)

  println("inner0 color: ", colors(report.toplevel.inner0.cd.registered.clockDomain))
  println("inner1 color: ", colors(report.toplevel.inner1.cd.registered.clockDomain))

  assert(clockDomains(1).clock == report.toplevel.inner0.cd.registered.clockDomain.clock)
  assert(clockDomains(1).reset == report.toplevel.inner0.cd.registered.clockDomain.reset)
}
```

If you go through all the ClockDomains in the beginning then you can decide on colors/names etc., store them in the map and use where needed.

andreasWallner, 2023-04-20:
And: I'd agree that it would be quite confusing to have multiple external CDs with the same name. In the examples I used them since it was a bit easier, but you have the same behavior with internal CDs - and there it's not much of an issue if you have multiple with the same name. (Since you anyhow never refer to them with the name)

ZhaokunHu, 2023-04-21:
Thank you for explaining. Now I understand how to distinguish between two CDs with the same name, and I have rebuilt my drawing code using objects instead of names.
[image]

closed 2023-04-21 by ZhaokunHu

andreasWallner, 2023-04-22:
@Readon You might have a point there, especially when considering that e.g. in the leaf components the clock is never connected to anything, for an extreme example:
```scala
object TestClocks extends App {
  class Inner() extends Component {
    val en = in port Bool()
    val i = in port Bool()
    val o = out port Bool()

    new ClockEnableArea(en) {
      o := RegNext(i)
    }
  }
  def makeInner(i: Bool, o: Bool, en: Bool) = {
    val inner = new Inner()
    inner.i := i
    o := inner.o
    inner.en := en
    inner
  }

  val report = SpinalVerilog(new Component {
    val i = in port Bits(6 bit)
    val o = out port Bits(6 bit)
    val en = in port Bits(4 bit)

    o(0) := RegNext(i(0))
    new ClockEnableArea(en(0)) {
      o(1) := RegNext(i(1))
    }
    val inner0 = makeInner(i(2), o(2), en(1))
    new ClockEnableArea(en(2)) {
      val inner = makeInner(i(3), o(3), en(3))
    }
    val inner1 = makeInner(i(4), o(4), en(3))
    new SlowArea(20 cycles) {
      o(5) := RegNext(i(5))
    }
  })

  HDElkDiagramGen(report)
}
```
which generates:
[image]
and for the inner components
[image]