SpinalHDL/SpinalHDL #1091: Continued inability to build on Windows - Java Exception in thread "main" java.lang.IllegalArgumentException

angelsix, 2023-04-19:
This has been going on for over a year now, when I work between PCs, Windows updates, or anything changes etc...

Uninstall everything (sbt, all javas, IntelliJ etc...)

Install sbt-1.8.2.msi
Install jdk-8u202-windows-x64.exe
Install ideaIC-2023.1.exe

Build a known working project in command line: sbt run

```
[info] running (fork) cc.CSVerilog
[info] [Runtime] SpinalHDL v1.6.4    git head : 598c18959149eb18e5eee5b0aa3eef01ecaa41a1
[info] [Runtime] JVM max memory : 14492.5MiB
[info] [Runtime] Current date : 2023.04.19 08:26:32
[info] [Progress] at 0.000 : Elaborate components
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 2.132 : Elaborate components
[error] Exception in thread "main" java.lang.IllegalArgumentException: Argument value 0xfffffffff exceeds native capacity (4 bytes) mask=0xffffffff00000000
[error]         at com.sun.jna.IntegerType.setValue(IntegerType.java:92)
[error]         at com.sun.jna.IntegerType.<init>(IntegerType.java:55)
[error]         at com.sun.jna.platform.win32.WinDef$DWORD.<init>(WinDef.java:135)
[error]         at net.openhft.affinity.impl.WindowsJNAAffinity.setAffinity(WindowsJNAAffinity.java:94)
[error]         at net.openhft.affinity.Affinity.setAffinity(Affinity.java:165)
[error]         at spinal.core.fiber.EngineContext.start(AsyncCtrl.scala:96)
[error]         at spinal.core.fiber.Engine$.create(AsyncCtrl.scala:150)
[error]         at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:2278)
[error]         at spinal.core.internals.PhaseContext.doPhase(Phase.scala:181)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$118.apply(Phase.scala:2581)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$118.apply(Phase.scala:2579)
[error]         at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2579)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2516)
[error]         at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
[error]         at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2516)
[error]         at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2511)
[error]         at spinal.core.Spinal$.apply(Spinal.scala:380)
[error]         at spinal.core.SpinalConfig.generate(Spinal.scala:162)
```

Can do this same step on 2 PCs and it works, can do it on 3 others, and it fails as above.

This was a fully working PC yesterday, installed some unrelated software and now this fails again. Wiped, removed all java, tested command line could not find Java, cleared all caches, and still same issue.

It completely stops the ability to build anything using SBT as the machines we are using are then totally incapable of building anything.

Any advise would be greatly appreciated.

Readon, 2023-04-19:
Maybe it's time to use docker image instead of setup environment every time.
I have created a base image for SpinalHDL can be used by 
``` shell
docker pull ghcr.io/readon/spinalhdl:master
```
May be, we can distribute one base image by each version releasing.

Dolu1990, 2023-04-19:
@angelsix How many CPU / Thread do you have ? seems like more than 18 cores, 36 threads ?

Seems like a net.openhft.affinity issue not expecting so many threads XD ?

Dolu1990, 2023-04-19:
Note that https://github.com/SpinalHDL/SpinalHDL/pull/1093 is likely fixing it.

angelsix, 2023-04-19:
> @angelsix How many CPU / Thread do you have ? seems like more than 18 cores, 36 threads ?
> 
> Seems like a net.openhft.affinity issue not expecting so many threads XD ?

Yeah 36 cores.

When can we get a public build with #1093 integrated?

Dolu1990, 2023-04-19:
could release a hotfix. Just to be sure all is good, can you try with a local release ? 
Following : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Introduction/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

angelsix, 2023-04-19:
Stuck at publish step

```
C:\Users\conta\Desktop\SpinalHDL>sbt clean '++ 2.11.12' publishLocal
[info] welcome to sbt 1.6.0 (Oracle Corporation Java 1.8.0_202)
[info] loading global plugins from C:\Users\conta\.sbt\1.0\plugins
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from C:\Users\conta\Desktop\SpinalHDL\project
[info] loading settings for project all from build.sbt ...
[info] set current project to SpinalHDL-all (in build file:/C:/Users/conta/Desktop/SpinalHDL/)
[success] Total time: 0 s, completed 19-Apr-2023 10:39:11
[error] Expected letter
[error] Expected symbol
[error] Expected '+'
[error] Expected '++'
[error] Expected 'java++'
[error] Expected 'java+'
[error] Expected '^'
[error] Expected '^^'
[error] Expected '+-'
[error] Expected 'debug'
[error] Expected 'info'
[error] Expected 'warn'
[error] Expected 'error'
[error] Expected 'addPluginSbtFile'
[error] Expected 'sbtRebootNetwork'
[error] Expected 'show'
[error] Expected 'print'
[error] Expected 'all'
[error] Expected 'Global'
[error] Expected '*'
[error] Expected 'Zero'
[error] Expected 'ThisBuild'
[error] Expected 'ProjectRef('
[error] Expected '{'
[error] Expected project ID
[error] Expected configuration
[error] Expected configuration ident
[error] Expected key
[error] Expected '__disconnectNetworkChannel'
[error] Expected '__waitWatch'
[error] Expected 'sbtPromptChannel'
[error] Expected end of input.
[error] Expected 'sbtRebootImpl'
[error] Expected 'early('
[error] Expected '-'
[error] Expected '--'
[error] Expected '!'
[error] Expected 'sbtReportResult'
[error] Expected 'sbtMapExec'
[error] Expected 'sbtCompleteExec'
[error] Expected '__failWatch'
[error] Expected '__preWatch'
[error] Expected '__postWatch'
[error] Expected '__runWatch'
[error] Expected '__stopWatch'
[error] '++
[error] ^
```

Dolu1990, 2023-04-19:
Seems like it doesn't like the quotes.

Can you try  : 
sbt clean "++ 2.12.13" publishLocal

Else directly : 
sbt clean publishLocal

angelsix, 2023-04-19:
Yep works now and builds

Dolu1990, 2023-04-21:
I'm now releasing 1.8.1a  Should be online in like 1h from now