SpinalHDL/SpinalHDL #1094: Support for VHDL Boolean type

corentin-fleury, 2023-04-20:
It would be nice to support the VHDL boolean type. Indeed as of now (1.8.0), if you have a Bool variable in SpinalHDL, it is translated to an std_logic once you generate VHDL code. One thing is that it might be confusing to have a Bool translate to std_logic and another type translate boolean.

Dolu1990, 2023-04-21:
Hi, 
Would supporting Boolean realy add value to the emited verilog ?

corentin-fleury, 2023-04-21:
I think it is something worth adding. I personally have a VHDL module which takes a Boolean as a generic and mapping it as Bool in Spinal makes it impossible to simulate using modelsim or synthesize in vivado

andreasWallner, 2023-04-22:
You write that your module takes a Boolean as a generic. Are you using it in SpinalHDL as a BlackBox? Would you have a small code sample that shows where you run into issues?
For synthesizable logic I always stuck with `std_ulogic` - would this also be a case for you, or only the generic?

Dolu1990, 2023-04-25:
> I personally have a VHDL module which takes a Boolean as a generic and mapping it as Bool

Boolean / Bool =>  in which context ? i mean, you mean Scala Boolean generic is mapped as a std_logic ?