SpinalHDL/SpinalHDL #1096: penable signal and the Psel signal will differ by 2 clock cycles after crossing the clock domain in apb3cctoggle

Amegodvivado, 2023-04-24:
Why is it that the penable signal and the Psel signal will differ by 2 clock cycles after crossing the clock domain in apb3cctoggle, which will not comply with the apb3 protocol, but according to the two signals I input, they are one cycle apart, in line with the regulations, is this reasonable?

Amegodvivado, 2023-04-27:
Can Apb3ccoggle only be used from slow clock domain to fast clock domain? What is the difference between Apbcc’s lib and Apbcctoggle?

Dolu1990, 2023-04-28:
Hi,

Can you share a wave file of the situation ?

Else there is Apb3CC bellow which is another implementation.

> Can Apb3ccoggle only be used from slow clock domain to fast clock domain? What is the difference between Apbcc’s lib and Apbcctoggle?

Maybe, i didn't implemented Apb3ccoggle, so i'm not sure the design.

Readon, 2023-05-01:
I have looked over the Apb3CCToggle a little bit. 
However, I could not agree with the design that directly use BufferCC on [multibits data on line 61](https://github.com/SpinalHDL/SpinalHDL/blob/703a7929225741fce6a6635cebfa59de6751dc3c/lib/src/main/scala/spinal/lib/bus/amba3/apb/Apb3CCToggle.scala#L62) and later lines.
May be a check on the usage of BufferCC might be valuable.

Dolu1990, 2023-05-03:
> May be a check on the usage of BufferCC might be valuable.

Hmm, like having to add a explicit tag or arguement to allow multibit bufferCC ?

typingArtist, 2023-05-03:
To me the whole implementation using BufferCC looks fishy. As APB can only handle one request at a time, the whole request needs to be transferred consistently to the target clock domain instead of its separate signals with individual CC pipelines, and the same applies to the return data and status.

An implementation should wait for a detected request (PENABLE && PSEL) in the inClk domain, then set the request payload (address, access type, write data) in a CC tagged FF once and use a PulseCCByToggle for the trigger. The outClk domain would detect the CC’d trigger and use the indicated request payload to rebuild a proper APB request in the outClk domain, starting with PENABLE, followed by PSEL, etc. All the outClk transaction handshake is performed by the APB master in the outClk domain. When the request is done, the return payload (read data, PSLVERR) is, again, written to a CC tagged FF, and the ack is returned through a PulseCCByToggle to the inClk domain. Once the CC’d ack is detected there, the return payload is fed to the inClk return signals and the request is finished.

andreasWallner, 2023-05-03:
@Dolu1990 This goes a bit beyond the scope of this issue, but: I currently see two possible problems with BufferCC. First is that it can easily be used for data that should stay cohorerent. Second is that it is unsafe in the sense that for a safe single-bit transition you need to guarantee that there is a register in the launching domain.
I wanted to open an issue to:
- provide two additional sync components: one for SingleBitCDC (a register + BufferCC) and one for non-coherent multibit (a bunch of registers + BufferCC)
- keep BufferCC as a low-level building block, but warn about it in the documentation

I don't think that a seperate tag would help much here: every use-case where it's OK to use BufferCC for more than one bit would have/need it anyways.

Looking at this specific case I'm not sure you need to (actually must not) synchronize the data at all. Synchronizing the control signals and just registering the others when those control signals are synchronized would be the way to go IMO. (ADDR e.g. has the same coherency issue.)

Dolu1990, 2023-05-03:
@andreasWallner 

Yes i agree.

> keep BufferCC as a low-level building block, but warn about it in the documentation

Yes

Good ^^

Another thing which could also be usefull is to add a tag "cross clock timing constraint". saying : that cross clock path should not be longer than X cycles.
That can then be used to generate SDC constraints.
In other words, documenting timing constraints in the hardware description for loater SDC automated generation.

@typingArtist So should we replace : https://github.com/SpinalHDL/SpinalHDL/blob/85e83a7b3553ec5e91cfb47f13539a9384a4d2b6/lib/src/main/scala/spinal/lib/bus/amba3/apb/Apb3CCToggle.scala#L41 
By : 
https://github.com/SpinalHDL/SpinalHDL/blob/85e83a7b3553ec5e91cfb47f13539a9384a4d2b6/lib/src/main/scala/spinal/lib/bus/amba3/apb/Apb3CCToggle.scala#L136

(seems like we should)

typingArtist, 2023-05-03:
@Dolu1990 yes, blind me lazily didn’t scroll down to see that there is almost an exact implementation of my suggestion already there! :smile: 

However, when we are at it, Apb3CC could benefit from a simplification, which might even be a fix. I’d make the two unidirectional channels Flows and use FlowCCByToggle for each of them, instead of a Stream in the downstream direction. There is a state boolean already there in both clock domains, and having two acknowledgments running upstream (one for the ready in the StreamCCByToggle, one in the valid in the FlowCCByToggle), racing each other, doesn’t make me sleep well considering consistency. The suggested improvement should reduce register count and LOCs as well.

typingArtist, 2023-05-03:
@Dolu1990  That sounds quite reasonable and nice!

Dolu1990, 2023-05-03:
Done in https://github.com/SpinalHDL/SpinalHDL/commit/f447738442c93650ba5eb385fa0d2c68371eab32

andreasWallner, 2023-05-03:
@Dolu1990 I split my comment above into a separate issue: #1109

Amegodvivado, 2023-05-04:
Thanks to the seniors for answering，for the latest version not being able to update in the server for the time being, when I pulled out APB3.scala and APB3cctoggle Flow.scala in lib and some others, trying to replace it with the latest code on github, some problems arose. Is there a better way to try out the latest code?

andreasWallner, 2023-05-04:
@Amegodvivado I'd recommend the publish to you local sbt cache, have a look here: https://spinalhdl.github.io/SpinalDoc-RTD/dev/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

Amegodvivado, 2023-05-05:
@andreasWallner Think you for your recommendation，but no internet  in the server now and the spinal verson is 1.6.4 and there are no plans for an update。I  try to  copy the new code in github to a new scala file and  import it in  my project(Apb3,Apb3CCtoggle,Flow,Clockdomian,) but  there are some problem. Is this the right approach?

Dolu1990, 2023-05-05:
@Amegodvivado I think you can use the Apb3CC from 1.6.4 directly. no need of update. maybe ?

Amegodvivado, 2023-05-05:
when I pulled out  APB3cctoggle in the 1.6.4 lib to my project.There may be some problems with this that do not compile：

https://github.com/SpinalHDL/SpinalHDL/blob/85e83a7b3553ec5e91cfb47f13539a9384a4d2b6/lib/src/main/scala/spinal/lib/bus/amba3/apb/Apb3CCToggle.scala#L173 ..`  
val outputCmd = StreamCCByToggle(inputLogic.inputCmd, inputClock, outputClock)   
 io.output.PENABLE := False
    io.output.PSEL := 0
    io.output.PADDR := outputCmd.PADDR
    io.output.PWDATA := outputCmd.PWDATA
..........
`This error is displayed：Cannot resolve overloaded method 'StreamCCByToggle'
 Cannot resolve symbol PADDR PWDATA PWRITE...

It seems that these signals such as PADDR cannot be found.

Amegodvivado, 2023-05-08:
@Dolu1990 Thanks, this issue has been resolved,it's the problem of the StreamCCByToggle method  overloaded.I try to pull out the object and rename it，then succefully!

closed 2023-05-08 by Dolu1990