SpinalHDL/SpinalHDL #1104: VHDL Blackbox generates component with unconstrained signals

C-Elegans, 2023-05-01:
I have a spinal component that instantiates a VHDL blackboxed component. When I generate the VHDL for these components and try to simulate it with GHDL, it gives me errors because the generated VHDL component declaration has no constraints for the width of the IO in the component.

For example, the following SpinalHDL blackbox:
```scala
case class Example() extends BlackBox {
  val io = new Bundle {
    val data_in = slave Flow(Vec(SInt(16 bits), 2))
    val data_out = out SInt(16 bits)
 }
  noIoPrefix()
  addTag(noNumericType)
}
```

When instantiated in another spinal module and VHDL is generated creates the following component declaration:

```VHDL
  component pe_single_path_spinal_wrapper is
    port( 
      io_data_in_0 : in std_logic_vector;
      io_data_in_1 : in std_logic_vector;
      io_data_in_2 : in std_logic_vector;
      io_data_in_3 : in std_logic_vector;
      io_data_out : out std_logic_vector
 );
```
where the std_logic_vectors have no width specified. When I attempt to simulate (or synthesize) this with GHDL I get errors like the following where the component is actually instantiated:
```
/path/to/file.vhd: type of actual conversion must be fully constrained
/path/to/file.vhd: type of formal conversion must be fully constrained
```

Dolu1990, 2023-05-03:
Ahhh, it maybe due to some convertion from Signed to std_logic_vector happening in the instance definition ? (noNumericType) ?
Guess there is no issue if you don't use noNumericType ? (i mean, still the issue that you need noNumericType, but i mean, from a GHDL perspective) ?

C-Elegans, 2023-05-03:
Yep, it appears to work correctly if I modify the blackbox/instantiation to work without noNumericType

Dolu1990, 2023-05-05:
Should be good with the dev branch now.

Let's me know if you have any issue

commit 6390bc9d (2023-05-05): fix #1104 VHDL blackbox with noNumericType
closed 2023-05-05 by Dolu1990