SpinalHDL/SpinalHDL #1108: UartCtrlRx: bitTimer.reset() to center bit in Majority Voter  PR#1107

dlmiles, 2023-05-02:
[image]
[image]
[image]

dlmiles, 2023-05-02:
VCD/FST for Image 1

[SpinalHDL_UART_IssueBefore.fst.zip](https://github.com/SpinalHDL/SpinalHDL/files/11369512/SpinalHDL_UART_IssueBefore.fst.zip)

VCD/FSR for Image 2

[SpinalHDL_UART_IssueAfter.fst.zip](https://github.com/SpinalHDL/SpinalHDL/files/11369513/SpinalHDL_UART_IssueAfter.fst.zip)

Simulation Scala to generate VCDs (setup for Image 3, but set NOISE_ENABLED=false for Image 1 and 2)

[UartSim.scala.txt](https://github.com/SpinalHDL/SpinalHDL/files/11369502/UartSim.scala.txt)

Dolu1990, 2023-05-03:
Hi,

Hmm i think the idea of the current implementation is to sample as much as possible in the middle of the sampler_value bit.
Which is what it  seems to do. The goal is to be as much as possible immune against jitter / clock drift. 
Why should it sample more toward the end of the bit ?

Thanks :)

dlmiles, 2023-05-03:
The diagrams show a RED circle where the commit changes the bitTimer.reset() value.  From 2 to 5.

The diagrams show the RED arrow with the contents of the MajorityVoter and a vertical RED line of where the sample is being taken in respect of the bit.  This is the influence range of samples inside the MajorityVoter.
This influence range is not at the center of the bit, it is biased to the leading side.

For timing purposes, the bitTimer.tick point is not the center of the bit in respect of sample.value (which is how it looks to have been implemented).
For timing purposes, the bitTimer.tick should point to the contents of the center sample of the MajorityVoter is the center of the bit.  Because this is the center of the samples that have the highest influence on sample.value at that time.

I agree concerning immunity against jitter / clock drift, the injection of noise case (3rd diagram) provides information into that area.  The 3rd diagram NOISE removes 2.5 sampler.ticks worth of signal from the leading side center of the bit.  There is actually 3 out of 8 samples removed, but the extra 0.5 occurs between 1 and 1.5 into 8 of the tick, which should be outside that center band of influence.  If you move the bitTimer.tick back in that scenario (to mimic 1st diagram) it would not have read the correct value.

I understand and agree with the points you are stating (in the issue comment above), the comments in the code indicate this understanding as well, but this information does not relate to how the existing implementation works.

andreasWallner, 2023-05-03:
I had a look again at the implementation, I might be restating the obvious, but humor me for a second... my 0.02$ (and I'll use the concrete values from the config that you used):

As a recap, the config is for
- 1 pre sample
- 5 samples
- 2 post-samples

Which I read that we want to have 8 samples per bit. Of those 8 bit we want to ignore the first 1 sample and the last 2 samples, only looking at the 5 "center" samples. 
To only need to have a single counter the implementation tricks a bit: it shifts the bitgrid (`bitTimer.tick`) such that it occurs at the point where we want to look at the majority voter content, in our case this is 6 samples after the start of the bit (the one we want to ignore has been shifted out of the majority voter at that point in time).

The start-bit detection uses the majority voter, it therefore has a delay with the start bit edge (ceil(sample/2) = 3 cycles).
We have one cycle delay because we only look at the majority voter output at the next `sampler.tick`. To get the wanted offset of 6 cycles into the bit we need to set the counter to 6-3-1=2 cycles.

[image]

To your example with the noise: In that example it looks like our 8 samples are: 1 high, 3 low, 4 high. With the configuration that is set up where we ignore the first 1 sample, take the next 5 (3 low, 2 high) getting the bit wrong is to be expected. If this is what you want to protect against in your system I'd rather change the configuration accordingly, i.e.:
- 3 pre samples
- 5 samples
- 0 post samples

dlmiles, 2023-05-04:
[image]

Annotating your VCD image.  Would this be accurate with where each sample comes from ?

Not trying to make the noise example work, it is just an example of how much noise was needed to be introduced to get a change in behaviour.  Contrived example to help understand the implementation better.

The main protection concern is against a % clock difference of the transmitter.  Most reports I can find seem to relate to scenario when the TX being slightly too fast.  A bias to the leading side (pre=2,post=1) is also better for that.  Shifting tolerance ranges in favour of handling the (maybe) more likely scenario of tx too fast, possibly looking like -1.25% (tx slow) to +2.5% (tx fast) - for 10 bits.  I expect real numbers to be much better than this due to MajorityVoter use / width.

andreasWallner, 2023-05-04:
The annotations look correct to me.

Ad your example: I understand, I did pretty much the same thing the first time I looked at it ;-)

dlmiles, 2023-05-10:
Thanks for the review on this.  I am agreeing with your analysis.

I shall work in confirming error rates with testbenches designed to introduce signal error, before then making an ASIC out of it :)

closed 2023-05-10 by dlmiles