SpinalHDL/SpinalHDL #1110: Unconditional assignment generated when forcing existence of signal that would be pruned

andreasWallner, 2023-05-06:
When looking at a question of @likewise I stumbled over this "gem"...
If I force a signal that is created in a conditional context, would describe a latch and would also be pruned to exist (by giving it a proper name) then it is created as an unconditionally assigned signal and also listed as having been pruned.
```scala
object Repro extends App {
  SpinalVerilog(new Component {
    val sel = in port Bool()
    val something = in port Bits(4 bit)
    when(sel) {
      val error = something.resize(5)
      error.setName("error")
    }
  }).printPruned()
}
```

Generates:
```verilog
module unamed (
  input               sel,
  input      [3:0]    something
);

  wire       [4:0]    error;

  assign error = {1'd0, something};

endmodule
```
and outputs
```
...
[Progress] at 0.649 : Generate Verilog
[Warning] 1 signals were pruned. You can call printPruned on the backend report to get more informations.
[Done] at 0.751
[Warning] Pruned wire detected : (toplevel/error :  Bits[5 bits])
```

I'm honestly not sure what I'd expect the outcome to be here, most likely an error that there is a latch in the code? Or is this expected behavior?

Dolu1990, 2023-05-07:
Hi,

That behaviour is the one i'm expecting.

error.resized(5) is a function which will read the final value of the error signal (unconditionaly) and resize it to 5 bits. it doesn't care about the conditional statement it is in.

The reason is : 

```scala
  SpinalVerilog(new Component {
    val sel = in port Bool()
    val something = in port Bits(4 bit)
    val result = out port Bits(5 bit)
    result  := 0
    when(sel) {
      result := something.resize(5)
    }
  }).printPruned()
````

In that example, you kinda only want the "when" to affect the result assignement, not the expression on the right operand

andreasWallner, 2023-05-13:
Understood, wasn't considering that spinal can't really act on the `=` operator there, like it would with `:=`.
It might still make sense to track this and see if there are any other assignments to `error` to still be able to generate an error - I'll close this and give it some thought when I have some more time.

closed 2023-05-13 by andreasWallner