SpinalHDL/SpinalHDL #1111: Question About The implicit function of DataCarrier

yuqi-ali, 2023-05-09:
look this code:
```
case class StreamTest() extends Component{
  val io=new Bundle{
    val port_in=slave(Stream(Fragment(Fragment(UInt(8 bits)))))
    val fire0,fire1=out Bool()
  }
  io.port_in.ready.set()
  io.fire0:=io.port_in.lastFire
  io.fire1:=io.port_in.fire&io.port_in.last
}
```
the RTL Generated is:
```
module StreamTest (
  input               io_port_in_valid,
  output              io_port_in_ready,
  input               io_port_in_payload_last,
  input               io_port_in_payload_fragment_last,
  input      [7:0]    io_port_in_payload_fragment_fragment,
  output              io_fire0,
  output              io_fire1
);

  wire                io_port_in_fire;
  wire                io_port_in_fire_1;

  assign io_port_in_ready = 1'b1;
  assign io_port_in_fire = (io_port_in_valid && io_port_in_ready);
  assign io_fire0 = (io_port_in_fire && io_port_in_payload_last);
  assign io_port_in_fire_1 = (io_port_in_valid && io_port_in_ready);
  assign io_fire1 = (io_port_in_fire_1 && io_port_in_payload_fragment_last);

endmodule
```
fire0 is not equal to fire1.
why it call different implicit function in DataCarrier(toImplicit,toImplicit2)?

yuqi-ali, 2023-05-10:
I'm not focus on the Stream.fire。
io_fire0 use the io_port_in_payload_last,
io_fire1 use the io_port_in_payload_fragment_last.
As the definition of lastFire: `def lastFire : Bool = pimped.fire && pimped.last`,why it call the toImplicit,not the toImplicit2(io.port_in.last call the toImplicit2 seems ok)?

Readon, 2023-05-10:
It seems that [reply](https://github.com/SpinalHDL/SpinalHDL/issues/955#issuecomment-1311627303) in #955 can be a good answer. The Fragment[T] => T so that Stream[Fragment[Fragment[T]]] => Stream[Fragment[T]] => Stream[T].

Dolu1990, 2023-05-10:
Hoooo i see now.

So, seems like in scala, when there is multiple function prototype which could match, scala choose the one which is the most precise with the type.
In other words, DataCarrier[Fragment[T]] is closer than DataCarrier[T] to the stream definition

A simple workaround is to add  : 

  implicit def toImplicit3[T <: Bundle](dataCarrier: DataCarrier[Fragment[Fragment[T]]]): Fragment[T] = dataCarrier.payload

closed 2023-05-10 by Dolu1990

yuqi-ali, 2023-05-11:
Yea
In this example,DataCarrier[Fragment[T]] is closer than DataCarrier[T] to the stream definition.But Why the lastFire in the DataCarrierFragmentPimped use the  DataCarrier[T] as it is closer to DataCarrier[Fragment[T]] ?
I also test as follow:
```
case class StreamTest() extends Component{
  val io=new Bundle{
    val port_in=slave(Stream(Fragment(Fragment(UInt(8 bits)))))
    val fire0,fire1=out Bool()
  }
  def lf[T <: Data](port:Stream[Fragment[T]]):Bool={
    port.fire & port.last
  }
  io.port_in.ready.set()
  io.fire0:=lf(io.port_in)
  io.fire1:=io.port_in.last & io.port_in.fire
}
```
the fire1 use the DataCarrier[Fragment[T]] as expected, but while use the DataCarrier[T] if i write the same logic in a fucntion(lf) and use this function。A little confused~
 It's rare for me to use  Stream(Fragment(Fragment[T]) in my daily work, just for studing.

yuqi-ali, 2023-05-11:
I also test as follow:
```
case class port() extends Bundle{
  val last=Bool()
}
case class StreamTest1() extends Component{
  val io=new Bundle{
    val port_in=slave(Stream(Fragment(port())))
    val fire0,fire1=out Bool()
  }
  noIoPrefix()
  def lf[T <: Data](port:Stream[Fragment[T]]):Bool={
    port.last
  }
  io.port_in.ready.set()
  io.fire0:=io.port_in.last
  io.fire1:=lf(io.port_in)
}
```
fire0,fire1 also call different implicit function.

commit 3519cef1 (2023-05-09): fix #1111 by using signalCache on Stream.fire
commit 8fc0a92c (2023-05-10): Fix #1111 DataCarrier[Fragment[Fragment[T]] payload implicit