SpinalHDL/SpinalHDL #1116: a null

MindShow, 2023-05-11:
```scala
case class MyIOA(xEn:Boolean = false) extends BlackBox {
  val io = new Bundle{
    val x = in(Bool()).genIf(xEn)
    val y = in(Bool())
  }
  addPrePopTask(() => {
    io.flattenForeach(port => {
      println(port.name)
    })
  })
}

case class MyIOB(xEn: Boolean = false) extends Component {
  val io = new Bundle {
    val x = in(Bool()).genIf(xEn)
    val y = in(Bool())
  }

}

case class MyIOTest(xEn:Boolean = false) extends Component {
  val io = new Bundle{
    val aIo = new Bundle {
      val x = in(Bool()).genIf(xEn)
      val y = in(Bool())
    }
    val bIo = new Bundle {
      val x = in(Bool()).genIf(xEn)
      val y = in(Bool())
    }
  }
  val myIoA= MyIOA()
  val myIoB= MyIOB()
  io.aIo <> myIoA.io
  io.bIo <> myIoB.io
}

object MyMain{
  def main(args: Array[String]){
    SpinalConfig().generateVerilog(MyIOTest())
  }
}```

MindShow, 2023-05-11:
# this can work
```scala
case class MyIOA(xEn:Boolean = false) extends BlackBox {
  val io = new Bundle{
    val x = if(xEn) in(Bool()) else new BoolPoison
    val y = in(Bool())
  }
  addPrePopTask(() => {
    io.flattenForeach(port => {
      println(port.name)
    })
  })
}

case class MyIOB(xEn: Boolean = false) extends Component {
  val io = new Bundle {
      val x = in(Bool()).genIf(xEn)
    val y = in(Bool())
  }

}

case class MyIOTest(xEn:Boolean = false) extends Component {
  val io = new Bundle{
    val aIo = new Bundle {
      val x = if(xEn) in(Bool()) else new BoolPoison
      val y = in(Bool())
    }
    val bIo = new Bundle {
      val x = in(Bool()).genIf(xEn)
      val y = in(Bool())
    }
  }
  val myIoA= MyIOA()
  val myIoB= MyIOB()
  io.aIo <> myIoA.io
  io.bIo <> myIoB.io
}

object MyMain{
  def main(args: Array[String]){
    SpinalConfig().generateVerilog(MyIOTest())
  }
}
```

MindShow, 2023-05-11:
# or this
```
case class MyIOA(xEn:Boolean = false) extends BlackBox {
  val io = new Bundle{
    val x = xEn generate in(Bool())
    val y = in(Bool())
  }
  addPrePopTask(() => {
    io.flattenForeach(port => {
      println(port.name)
    })
  })
}

case class MyIOB(xEn: Boolean = false) extends Component {
  val io = new Bundle {
      val x = xEn generate in(Bool())
    val y = in(Bool())
  }

}

case class MyIOTest(xEn:Boolean = false) extends Component {
  val io = new Bundle{
    val aIo = new Bundle {
      val x = if(xEn) in(Bool()) else new BoolPoison
      val y = in(Bool())
    }
    val bIo = new Bundle {
      val x = in(Bool()).genIf(xEn)
      val y = in(Bool())
    }
  }
  val myIoA= MyIOA()
  val myIoB= MyIOB()
  io.aIo <> myIoA.io
  io.bIo <> myIoB.io
}

object MyMain{
  def main(args: Array[String]){
    SpinalConfig().generateVerilog(MyIOTest())
  }
}
```

closed 2023-05-11 by MindShow

Readon, 2023-05-11:
I have checked code about genIf function, it seems that it is a deprecated one. How about give out a warning while it is used?
The latest usage of genIf is [here](https://github.com/SpinalHDL/SpinalHDL/blob/4a56ace222042cd4ca1e8dd1f8ab459b598859b0/lib/src/main/scala/spinal/lib/blackbox/lattice/ice40/Blackbox.scala#L341) which might be affected if deprecated warning is added.
@andreasWallner do you have any concern about this?

andreasWallner, 2023-05-13:
@Readon I checked, both the option most used in the codebase
```scala
case class TestBundle(s: Boolean) extends Bundle {
  val x = Bool()
  val y = if(s) Bool() else null
}
``` 
and the one that I have been playing around with lately
```scala
case class TestBundle(s: Boolean) extends Bundle {
  val x = Bool()
  val y = if(s) Some(Bool()) else None
}
```
don't have the same problem, didn't know about that when I used `genIf` for the Blackbox.
I'll open a PR that changes the use in the PLL and adds a deprecation notice to `genIf`

@MindShow Thanks for bringing this up, just as a note: there should not be a need to use `BoolPoison` - the version with `null` should work.

Readon, 2023-05-13:
> 

There is a ```generate``` function of Boolean type could do the same thing and is widely adopted.

Readon, 2023-05-13:
It might be more useful to have it described in the documents.