SpinalHDL/SpinalHDL #1117: How

MindShow, 2023-05-12:
请教 blackbox里的没有用到的 io 信号，该如何处理呢

case class BlackBox_T() extends BlackBox{
  val axis = slave(Axi4(Axi4Config(
    addressWidth = 30,
    dataWidth    = 512,
    idWidth      = 4,
    arUserWidth  = 1,
    awUserWidth  = 1)))

  axis.ar.addr   .setName("axis_ar_addr")
  axis.ar.ready  .setName("axis_ar_ready")
  axis.ar.payload.id  //如果verilog里的module没有用到这个信号，该如何处理呢

}

Dolu1990, 2023-05-12:
So, SpinalHDL will always check that all inputs of blackboxes and all output of the toplevel are driven by something.

That's a safety check.
You can else do axis.ar.id default(0) i think

Dolu1990, 2023-05-12:
Hmm can you provide a full self contained example for me to reproduce ?

About id = null, if the axi configuration has useId = false it will do it

MindShow, 2023-05-12:
# test code

```
case class AXI_IO() extends Bundle{
  val AA = in(Bool())
  val BB = in(Bool())
  val CC = in(Bool())
  val DD = out(Bool())
}

case class ADD_BB() extends BlackBox{
  val axi = AXI_IO()

  setBlackBoxName("axi_verilog")
  axi.AA.setName("A")
  axi.BB.setName("B")
  axi.CC.setName("C")    //Need remove
  axi.DD.setName("D")

  setInlineVerilog(
    """module axi_verilog
      |(
      |    input      A,
      |    input      B,
      |    output     D
      |);
      |
      |always @(*) begin
      |  D = A & B;
      |end
      |
      |endmodule
    """.stripMargin
  )
}

case class ADD() extends Component {
  val axi = AXI_IO()

  val bb = ADD_BB()
  axi <> bb.axi
}

object BlackBox_T_Main{
  def main(args: Array[String]){
    val config = SpinalConfig(
      targetDirectory = "PH1_T"
    )
    config.generateVerilog(ADD()).mergeRTLSource()
  }
}

```

# verilog

```
// Generator : SpinalHDL v1.8.1    git head : 2a7592004363e5b40ec43e1f122ed8641cd8965b
// Component : ADD
// Git hash  : 521f27563f03032872219ae7aa000b2bf8b4115b

`timescale 1ns/1ps

module ADD (
  input               axi_AA,
  input               axi_BB,
  input               axi_CC,     //need remove
  output              axi_DD
);

  wire                bb_D;

  axi_verilog bb (
    .A (axi_AA), //i
    .B (axi_BB), //i
    .C (axi_CC), //i
    .D (bb_D  )  //o
  );
  assign axi_DD = bb_D;

endmodule

module axi_verilog
(
    input      A,
    input      B,
    output     D
);

always @(*) begin
  D = A & B;
end

endmodule
    

```

andreasWallner, 2023-05-13:
This might be a dumb question, but when do you need this? The BlackBox which is some external component will either have the signal or not? If there is an option to be had then don't create the IO in the first place?

For you original example that would be:
```scala
case class BlackBox_T() extends BlackBox{
  val axis = slave(Axi4(Axi4Config(
    addressWidth = 30,
    dataWidth    = 512,
    useId        = false,
    arUserWidth  = 1,
    awUserWidth  = 1)))

  axis.ar.addr   .setName("axis_ar_addr")
  axis.ar.ready  .setName("axis_ar_ready")
}
```

For custom Bundle you can do a similar thing:
```scala
case class MyBundle(withX: Boolean) extends Bundle {
  val y = Bool()
  val x = if(withX) Bool() else null
}
```

Or did I misunderstand the problem?

MindShow, 2023-05-13:
different module use same bus protocol but whith just a little different when instance，
use custom bundle can really solve this problem, or wapper out blackox
then there Axi4_Custom1, Axi4_Custom2, ...

andreasWallner, 2023-05-13:
For AXI you shouldn't need a custom Bundle, it can be configured via parameters anyway. As shown above e.g. disabling the ID can simply be done via the `useId` parameter. I just added the custom bundle for completeness, not as a solution for AXI - sorry for causing confusion.
Is there some parameter missing in the `Axi4` that you'd want to be able to change that is currently not available?

MindShow, 2023-05-14:
thanks， maybe custom bundle is the best solution

closed 2023-05-14 by MindShow

andreasWallner, 2023-05-14:
@MindShow just as an afterthought: if you have AXI ports then you should not need any custom bundles (since the Axi is configurable) - if there is a configuration setting missing please let us know what it is.