SpinalHDL/SpinalHDL #1122: BusIf SystemRdlGenerator generates line feed as ‘\r\n’ in 'doc' field

cheerryio, 2023-05-18:
My test code is below:
```
package Test

import spinal.core._
import spinal.lib.bus.amba4.axilite._
import spinal.lib.bus.regif._
import spinal.lib.bus.misc._
import spinal.lib.bus.regif.AccessType._

class TestBusif() extends Component{
    val bus=AxiLite4(AxiLite4Config(addressWidth = 7, dataWidth = 32)).intoSlave()
    val busif=AxiLite4BusInterface(bus = bus, sizeMap = SizeMapping(0,4 KiB))
    val control=busif.newReg(doc="Core Control Register")
    val coreEnable=control.field(Bool(), RW, 0, doc=
        s"""Enable/Disable the core
           |0 - Stop generating packets
           |1 - Start generating packets
           |Controller ends the current transfer by resetting all internal FIFOs and registers.
           |Once enabled, controller start from VSS packet.(that is new video frame)
           |""".stripMargin)

    busif.accept(SystemRdlGenerator(
        fileName = "TestBusif",
        addrmapName = "TestBusif",
        name = Some("TestBusif"),
        desc = Some("generated systemRDL file")
    ))
}

object TestBusif{
    def main(args:Array[String]):Unit={
        SpinalConfig(
            mode=Verilog,
            defaultConfigForClockDomains=ClockDomainConfig(resetKind=SYNC,resetActiveLevel=HIGH),
            targetDirectory="verilog/Test",
            inlineRom=true
        ).generate(new TestBusif())
    }
}
```

There are several line feeds in the 'doc' field and the generated systemRDL file substitude the line feed with '\r\n'. As a result, my systemRDL parsing tool report me token reconition error. I have tried several systemRDL parsing tools like peakrdl and ordt. They all give me this error.

My current solution is to delete the '\r\n' in the generated file by hand and it works. But I think there is supposed to be a fix in the SystemRdlGenerator.

Readon, 2023-05-18:
Which platform are you using? If it is on Windows, Scala use "\r\n" as default line break for multi-line string.

cheerryio, 2023-05-18:
Yes, on windows.