SpinalHDL/SpinalHDL #1125: Bug with AxiLite4/Driver

allexoll, 2023-05-22:
I think I've stumbled upon a bug with the AxiLite4/AxiLite4SlaveFactory/AxiLite4Driver, where depending the run, a transaction would stall for no reason.

Here is an minimal code example:
```Scala
class MinimumCodeExample() extends Component{
  val io = new Bundle{
    val input = in Bool()
    val output = out Bool()
    val axilite = slave(AxiLite4(4, 32))
  }
  val busCtrl = AxiLite4SlaveFactory(io.axilite)
    io.output := False
    busCtrl.read(io.input, 0x00)
    busCtrl.write(io.output, 0x04)
}

object mceSim extends App{
  Config.sim.withFstWave.compile(new MinimumCodeExample).doSim { dut =>
    // Fork a process to generate the reset and the clock on the dut
    dut.clockDomain.forkStimulus(period = 10)
    dut.clockDomain.waitActiveEdge(10)
    val driver = AxiLite4Driver(dut.io.axilite, dut.clockDomain)
    driver.write(0x00, 0x0eadbeef)
    driver.read(0x04)
    dut.clockDomain.waitActiveEdge(10)
  }
}
```

Depending on the run, the simulation either

allexoll, 2023-05-22:
Same bug as #1024 but the fix was only made for regIf

andreasWallner, 2023-05-23:
I only had a quick look, but seeing how `AxiLite4Driver` is implemented it looks like you need to call `reset()` otherwise the `b.ready` line will not be driven during the simulation. On verilator it will keep its random reset value and give you a 50% chance that the simulation gets stuck.

Readon, 2023-05-24:
> I only had a quick look, seeing how `AxiLite4Driver` is implemented it looks like you need to call `reset()` otherwise the `b.ready` line will not be driven during the simulation. On verilator it will keep its random reset value and give you a 50% chance that the simulation gets stuck.

I have also searched for this reset function before, that's why I think #1006 might be helpful.

andreasWallner, 2023-05-24:
@Readon For that case I'd rather argue that the driver should call the reset function at instantiation time (like most others do).
On the topic of #1006: Not sure it makes sense to try to change how verilator works...

kleinai, 2023-05-24:
I should chime in because I added the offending line.  The idea was to have write block to match the behavior of read, after the command returns the write is complete.
Although yes, those signals should be reset and defined when the simulation starts much like how StreamDriver works.

Readon, 2023-05-24:
> @Readon For that case I'd rather argue that the driver should call the reset function at instantiation time (like most others do) 

Yes, I also used the reset function after a long search. So I hope it will ease the pain for other people who want to use it. I checked the documentation about Simulation again and there is no description about reset and Agent yet. Their behavior pattern is also different from StreamDriver which might lead misunderstand easily.

- I don't see a reason why it shouldn't. On the topic of #1006: Not sure it makes sense to try to change how verilator works - but that's a discussion for another issue ;-)

If #1006 is solved, at least we could know that which signals are not assigned, and that might lead errors. 
I don't think it's necessary to change the verilator, but check the assignments in spinal's sim lib.

allexoll, 2023-05-25:
Looking around for other examples, so as we can be coherent troughout the project, I found this [bmbDriver](https://github.com/SpinalHDL/SpinalHDL/blob/180eabc827e49ba58d55115b8c21cddc0b08716f/lib/src/main/scala/spinal/lib/bus/bmb/sim/BmbDriver.scala#L13) where as part of the init, some signals are set to a defined value. 

I think that should be the case for all.
As soon as the Driver is created with the bus as a parameter, the driver should "own" the bus. That might be a smaller fix than #1006 , and avoid a lot of issues for future users. If everyone is fine with this, I can do a PR in the comming days to close this (essentially just moving the `.reset` in the body of the Driver.

Readon, 2023-05-27:
> Looking around for other examples, so as we can be coherent troughout the project, I found this [bmbDriver](https://github.com/SpinalHDL/SpinalHDL/blob/180eabc827e49ba58d55115b8c21cddc0b08716f/lib/src/main/scala/spinal/lib/bus/bmb/sim/BmbDriver.scala#L13) where as part of the init, some signals are set to a defined value.
> 
> I think that should be the case for all. As soon as the Driver is created with the bus as a parameter, the driver should "own" the bus. That might be a smaller fix than #1006 , and avoid a lot of issues for future users. If everyone is fine with this, I can do a PR in the comming days to close this (essentially just moving the in the body of the Driver.`.reset`

Yes, initiate signals to default value while construction could be a way. 

To be more generic, is that reasonable to call a reset function for all XXXXDriver's construction? 
Then does a unified BusDriver interface is good practice for all logic that run on master end of bus?
At least, while creating a new XXXDriver some necessary functions would not be missed.

andreasWallner, 2023-05-29:
@Readon it's not only the bus drivers, also Stream, Flow, etc. - not sure we need a common interface for something that pretty much all drivers do without having to call a function.

Dolu1990, 2023-05-29:
>  not sure we need a common interface for something that pretty much all drivers do without having to call a function.

Agree

Readon, 2023-05-29:
> @Readon it's not only the bus drivers, also Stream, Flow, etc. - not sure we need a common interface for something that pretty much all drivers do without having to call a function.

It might be mixed.
1. call reset function in default construction might be useful.
2. other function of bus driver can have a unified interface, for example have a ```write``` and ```read``` function.

Do you mean item 1 is not that necessary or 2? :)

andreasWallner, 2023-06-02:
@allexoll While we were getting side-tracked - did you try if calling `reset()` at the start of the simulation resolves your issue?

andreasWallner, 2023-06-02:
@Readon 
Ad reset function: as noted above this should at least be called during construction - I think we are in agreement here.

Ad common read/write: Looking at what we currently have and how different the feature sets of busses are (e.g. compare APB vs. TileLink) I'm not sure it would be very useful, apart from the most simple read/write cases?

allexoll, 2023-06-03:
Yes!

PR #1139 (2023-06-12): Reset AxiLite4Driver on instantiation
Closes #1125

# Context, Motivation & Description

The AxiLite4Driver's behavior differs from most other drivers that it needed a call to `.reset()` to drive some static, necessary levels. Call reset on instatiation to remove this confusion, keep reset to not break old code. Remove unnecessary `reset()` calls in tests so that they are not bad examples.

# Impact on code generation

none
closed 2023-06-03 by allexoll