SpinalHDL/SpinalHDL #1127: Analog signal connect error with blackbox

zyn810039594, 2023-05-24:
Here's the blackbox:`class W25Q128JVxIM extends BlackBox {
    val io = new Bundle {
        val CSn   = in Bool ()
        val CLK   = in Bool ()
        val DIO   = inout(Analog(Bool()))
        val DO    = inout(Analog(Bool()))
        val WPn   = inout(Analog(Bool()))
        val HOLDn = inout(Analog(Bool()))
    }
    addRTLPath("./tmp/qspi_init.v")
    addRTLPath("./sim/model/qspi_model/W25Q128JVxIM.v")
    noIoPrefix()
}`
And here's the connection:`val model = new W25Q128JVxIM
    model.io.CSn:=io.CSn
    model.io.CLK:=io.CLK
    io.DIO.read:=model.io.DIO
    when(io.DIO.writeEnable) {
        model.io.DIO:=io.DIO.write
    }
    io.DO.read := model.io.DO
    when(io.DO.writeEnable) {
        model.io.DO := io.DO.write
    }
    io.WPn.read := model.io.WPn
    when(io.WPn.writeEnable) {
        model.io.WPn := io.WPn.write
    }
    io.HOLDn.read := model.io.HOLDn
    when(io.HOLDn.writeEnable) {
        model.io.HOLDn := io.HOLDn.write
    }`
Generated Verilog code:
`W25Q128JVxIM model (
    .CSn   (io_CSn), //i
    .CLK   (io_CLK), //i
    .DIO   ({analog_wrap_0[0]}),
    .DO    ({analog_wrap_1[0]}),
    .WPn   ({analog_wrap_2[0]}),
    .HOLDn ({analog_wrap_3[0]}) 
  );
  assign analog_wrap_0[0] = _zz_analog_wrap_0 ? io_DIO_write : 1'bz;
  assign analog_wrap_1[0] = _zz_analog_wrap_1 ? io_DO_write : 1'bz;
  assign analog_wrap_2[0] = _zz_analog_wrap_2 ? io_WPn_write : 1'bz;
  assign analog_wrap_3[0] = _zz_analog_wrap_3 ? io_HOLDn_write : 1'bz;
  always @(*) begin
    _zz_analog_wrap_3 = 1'b0; 
    if(io_HOLDn_writeEnable) begin
      _zz_analog_wrap_3 = 1'b1;
    end
  end
  always @(*) begin
    _zz_analog_wrap_2 = 1'b0; 
    if(io_WPn_writeEnable) begin
      _zz_analog_wrap_2 = 1'b1; 
    end
  end
  always @(*) begin
    _zz_analog_wrap_1 = 1'b0; 
    if(io_DO_writeEnable) begin
      _zz_analog_wrap_1 = 1'b1; 
    end
  end
  always @(*) begin
    _zz_analog_wrap_0 = 1'b0;
    if(io_DIO_writeEnable) begin
      _zz_analog_wrap_0 = 1'b1; 
    end
  end
  assign io_DIO_read = DIO; 
  assign io_DO_read = DO; 
  assign io_WPn_read = WPn; 
  assign io_HOLDn_read = HOLDn;`
Just look at the connection of read port,what they've connected.....

Dolu1990, 2023-05-24:
Hi,

Thanks for the issue ^^
I can reproduce with : 

```scala
class Xxx extends Component{
    val ctrl = slave (TriState(Bool()))
    
    val bb = new BlackBox {
      val data   = inout(Analog(Bool()))
    }

    ctrl.read := bb.data
    when(ctrl.writeEnable) {
      bb.data := ctrl.write
    }
}
```

I'm taking a look

Dolu1990, 2023-05-24:
Here is a workaround for now : 

```scala
    val ctrl = slave (TriState(Bool()))

    val bb = new BlackBox {
      val data   = inout(Analog(Bool()))
    }

    val tmp = Analog(Bool())

    ctrl.read := tmp
    when(ctrl.writeEnable) {
      tmp := ctrl.write
    }
```

Not great, i will look for a proper fix.

zyn810039594, 2023-05-24:
Thanks for you reply!
But,
> ```scala
>     val ctrl = slave (TriState(Bool()))
> 
>     val bb = new BlackBox {
>       val data   = inout(Analog(Bool()))
>     }
> 
>     val tmp = Analog(Bool())
> 
>     ctrl.read := tmp
>     when(ctrl.writeEnable) {
>       tmp := ctrl.write
>     }
> ```
Is 'bb.data' connected?

Dolu1990, 2023-05-24:
ahhh my bad, forgot it XD

Here it is : 

```scala
    val ctrl = slave (TriState(Bool()))

    val bb = new BlackBox {
      val data   = inout(Analog(Bool()))
    }

    val tmp = Analog(Bool())
    tmp := bb.data
    ctrl.read := tmp
    when(ctrl.writeEnable) {
      tmp := ctrl.write
    }

```