SpinalHDL/SpinalHDL #1129: Spinal Sim problem :assertion failed: Verilator C++ model compilation failed

Mido-sun, 2023-06-01:
make: *** [/mingw64/share/verilator/include/verilated.mk:239：VMyTopLevel___024root__DepSet_h1d291712__0.o] 错误 1
make: *** 正在等待未完成的任务....
make: *** [/mingw64/share/verilator/include/verilated.mk:242：VMyTopLevel___024root__DepSet_h1d291712__0__Slow.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:242：VMyTopLevel__Syms.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:242：VMyTopLevel___024root__DepSet_h27682af4__0__Slow.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:242：VMyTopLevel__Trace__0__Slow.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:239：VMyTopLevel___024root__DepSet_h27682af4__0.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:239：VMyTopLevel__Trace__0.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:242：VMyTopLevel___024root__Slow.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:239：VMyTopLevel.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated_threads.o] 错误 1
make: *** [VMyTopLevel.mk:77：VMyTopLevel__spinalWrapper.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated_vcd_c.o] 错误 1
make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated.o] 错误 1

Mido-sun, 2023-06-01:
I don't know how to solve it at all

andreasWallner, 2023-06-02:
I'm sorry but there is not enough information here for most of us help.
What are you trying to do, what's happening that you don't expect, how can we reproduce the issue?

dlmiles, 2023-06-03:
```
find -name "verilator" -type d
cd simWorkspace/?????/

test -d verilator && echo CORRECT directory to continue || echo INCORRECT directory to continue

./verilatorScript.sh
## paste the errors seen
make -C verilator -f VMyTopLevel.mk VMyTopLevel CURDIR=$(pwd)
## paste the errors seen

egrep "public|module" -A20 -B1 */rtl/*  
## or just paste the external module() port interface and any simPublic items
```

Mido-sun, 2023-06-16:
> 

I will TRY TY

Mido-sun, 2023-06-16:
> 

this is happening when I try to run the sim template

dlmiles, 2023-07-27:
Any update on this information.

I think it is understood you have seen this problem occur when running a sim template.

What is needed to assist is more information about the environment, errors/warnings from verilator, the target project under test, the signals being accessed.