SpinalHDL/SpinalHDL #1131: Generated  redundant code

hongxiaoo, 2023-06-02:
When I define an array and assign it to a switch, redundant code will be generated.

```
  wire       [15:0]   _zz_1;
  wire       [15:0]   _zz_2;
  wire       [15:0]   _zz_3;

  assign _zz_1 = 16'h0;
  assign _zz_2 = 16'h0001;
  assign _zz_3 = 16'h0002;
```

```
class test ()  extends Component {
  val io = new Bundle {
    val x = in(SInt(16 bits))
    val o = out(SInt(16 bits))
  }
  noIoPrefix()
  var tablog = Array(0, 1455, 2866)
  switch(io.x) {
    for (i <- 0 until (3)) {
      is(i) {
        io.o := tablog(i)
      }
    }
    default {
      io.o := 0
    }
  }
}
```
The verilog code generated using SpinalHDL v1.8.1 is as follows ：
```
module test (
  input      [15:0]   x,
  output reg [15:0]   o
);

  wire       [15:0]   _zz_1;
  wire       [15:0]   _zz_2;
  wire       [15:0]   _zz_3;

  assign _zz_1 = 16'h0;
  assign _zz_2 = 16'h0001;
  assign _zz_3 = 16'h0002;
  always @(*) begin
    case(x)
      16'h0 : begin
        o = 16'h0;
      end
      16'h0001 : begin
        o = 16'h05af;
      end
      16'h0002 : begin
        o = 16'h0b32;
      end
      default : begin
        o = 16'h0;
      end
    endcase
  end
endmodule
```

Dolu1990, 2023-06-03:
Hi ^^
Hmm right, i will take a look for a fix.

Dolu1990, 2023-08-07:
Got it fixed, it wasn't what i was expecting XD

commit 1a5db50f (2023-08-07): Fix #1131
closed 2023-08-07 by Dolu1990