SpinalHDL/SpinalHDL #1132: VCS Simulation Error same as #669

MindShow, 2023-06-04:
same as [#669 ](https://github.com/SpinalHDL/SpinalHDL/issues/669)
edid sim/src/main/scala/spinal/sim/VCSBackend.scala will work

```scala
@@ -281,7 +281,7 @@ class VCSBackend(config: VCSBackendConfig) extends VpiBackend(config) {
       case _               => " "
     }
     val topUnits = waveFormat match {
-      case WaveFormat.FSDB => toplevelName + " __simulation_def"
+      case WaveFormat.FSDB => "./rtl/" + toplevelName + ".v ./rtl/__simulation_def.v"
       case _               => toplevelName
     }
```

andreasWallner, 2023-06-05:
In the issue you linked @HC153Yuanyuan resolved this by updating/reinstalling VCS. Do you have any idea whether this might be a VCS version issue or something similar - the original code seemed to work out in the end? I don't have a VCS available for testing myself.

Readon, 2023-06-05:
> same as [#669 ](https://github.com/SpinalHDL/SpinalHDL/issues/669) edid sim/src/main/scala/spinal/sim/VCSBackend.scala will work
>
> ```scala
> @@ -281,7 +281,7 @@ class VCSBackend(config: [...]

As this patch is to add a searching directory 'rtl', it would be better to add inclusion of the dir in other way. By environment variables? 

Normally a fixed path is unpreferred to be added in the library.

MindShow, 2023-06-05:
SpinalHDL backend generate ```__simulation_def.v``` and put it in ```./rtl```  dir,  but vcs serch it in``` ./``` dir, 

or work as below
# SpinalHDL
```scala
@@ -281,7 +281,7 @@ class VCSBackend(config: VCSBackendConfig) extends VpiBackend(config) {
       case _               => " "
     }
     val topUnits = waveFormat match {
-      case WaveFormat.FSDB => toplevelName + " __simulation_def"
+      case WaveFormat.FSDB => " ./rtl/__simulation_def.v"
       case _               => toplevelName
     }
```
# App with ```-f filelist.f ```
```scala
  val flags = VCSFlags(
    compileFlags = List("-kdb "),
    elaborateFlags = List("-f filelist.f -kdb ")
  )
  val simConfig = SimConfig
    .withConfig(SpinalConfig(targetDirectory = "./"))
    .withVCS(flags)
    .withFSDBWave
```

wswslzp, 2023-06-05:
The code you refer to here is to generate **two top-level module names**, not two **file paths** of them.
Your change here may feed unexpected arguments to VCS compilation options and leads to compilation failures.

Pls check your VCS command line. Kindly remind that this may related to VCS version issue, as you referred in #669 .

MindShow, 2023-06-05:
# VCS INFO 
```shell
vcs script version : O-2018.09
machine name = fpga 
machine type = linux64 
machine os = Linux 5.19.0-43-generic  
The FLEXlm host ID of this machine is "00505690ffa6"
Compiler version = VCS O-2018.09-SP2_Full64
VCS Build Date = Feb 28 2019 22:34:30
```

# Solution
```scala
@@ -281,7 +281,7 @@ class VCSBackend(config: VCSBackendConfig) extends VpiBackend(config) {
       case _               => " "
     }
     val topUnits = waveFormat match {
-      case WaveFormat.FSDB => toplevelName + " __simulation_def"
+      case WaveFormat.FSDB => "-top " + toplevelName + " -top  __simulation_def"
       case _               => toplevelName
     }
 
```

# 
VCS backend use 3 steps simulation

MindShow, 2023-06-07:
SpinalHDL  work with 
**```VCS-MX O-2018.09-SP2_Full64```**
NOT
**``` VCS O-2018.09-SP2_Full64```**

closed 2023-06-07 by MindShow